mavlink/mavlink #1681: Camera Protocol really using gz compression?

olliw42, 2021-08-22:
The docs for the camera protocol says that the definition file may be gz compressed: https://mavlink.io/en/services/camera_def.html#file-compression.

I suspect that this should be changed to xz compression, as it is for COMPONENT_INFORMATION.

In my recollection, the idea to allow for compressed camera definition files is not so old, and subsequent work on COMPONENT_INFORMATION where initially also gz was suggested resulted in that xz is more suitable, and accordingly for COMPONENT_INFORMATION xz compression is now suggested. 

Given the history, I suspect that gz compression for camera definition files never actually made it into some actual "stable" implementation (I kind of think that I am the only one experimenting with this), so the change should be simple. 
If not, I wonder if the camera protocol should not also allow xz files, at least for mavftp uris. 
(I mean, one would have to bother with implementing both online gz and xz compression ... what a waste of resources)

Pl double-check the situation and clarify as needed.

:)

hamishwillee, 2021-08-22:
Camera stuff predates component information by years, and I have a feeling that the proposal for compression might have too. On the other hand since the requirement is on the GCS to support this, it an impossible change in requirements. 

@julianoes Do you know if anyone has implemented this?/ Have any objection to changing it? Worth chasing up with Gus?

olliw42, 2021-08-23:
> Camera stuff predates component information by years,

yes, sure, but concerning the compression it is indeed my recollection that it was me who suggested it while I was working on doing this stuff for STorM32 and that it thus can't be that old - my recollection might of course fool me

QGC I believe does not implement gz (again, my recollection is that I was doing a PR which never made it through, and I just checked few days ago its code and still can't see QGC doing it - again I might be wrong though)

whatever the history was, I think also camera should support xz now. If not instead of gz then at least in addition to gz.

hamishwillee, 2021-08-25:
@olliw42 Re https://github.com/mavlink/mavlink/issues/1681#issuecomment-903383322 ^^^ yes I think you did suggest compression. Yes it possibly isn't implemented/being used - even "probably". A question for @julianoes but I think this makes sense.

julianoes, 2021-08-30:
> @julianoes Do you know if anyone has implemented gz?/ Have any objection to changing it? Worth chasing up with Gus?

I don't think so. Let's change it to xz.

hamishwillee, 2021-09-01:
Done! The messages conveniently already said .xz :-)

olliw42, 2021-09-16:
appears to be fully resolved
so I close
THX

closed 2021-09-16 by olliw42