mavlink/mavlink #1682: MAV_TYPE: Shouldn't also MAV_AUTOPILOT be required tobe set correctly?

olliw42, 2021-08-23:
Concerning the heartbeat/connection protocol (https://mavlink.io/en/services/heartbeat.html), it describes nicely what one needs to know about MAV_TYPE.

However, in addition I think that also MAV_AUTOPILOT should be set to MAV_AUTOPILOT_INVALID if and only if the type is not an autopilot.

It is IMHO kind of logically required that MAV_AUTOPILOT is consistent with MAV_TYPE, and I therefore would expect the standard to require that it is set to MAV_AUTOPILOT_INVALID for non-autopilot components 

Besides logical consistency, it also would permit codes additional check for the type of a component. Pl note that from MAV_TYPE it is somewhat clumsy to check if something is an autopilot, and especially this is **_NOT_** a generic = future-proof method for doing so, since the list of autopilot MAV_TYPEs might be extended any time, so older code can't know. The MAV_AUTOPILOT flag would give a generic and future proof method to identify an autopilot.

I would suggest to put that to the standard (if it isn't already but just not mentioned) and to the docs.

hamishwillee, 2021-08-25:
The messages make this very clear, but as you say, this document does not. I've fixed in https://github.com/mavlink/mavlink-devguide/pull/382

In particular having a simple and reliable check for the autopilot component is a useful thing to note. Thanks very much.

olliw42, 2021-08-25:
GREAT!

many thx

closed 2021-08-25 by hamishwillee