mavlink/mavlink #1732: WIP tags in enums

python36, 2021-11-01:
Proposal to add WIP tag to enums. Some enums have comments that say it is a WIP enumeration, for example: AIS_FLAGS, FAILURE_UNIT. Some enums not have comments, but are only used in WIP messages, for example: CAMERA_TRACKING_STATUS_FLAGS, CAMERA_TRACKING_MODE, CAMERA_TRACKING_TARGET_DATA, GIMBAL_MANAGER_CAP_FLAGS, etc.

This issue is needed to justify the PR for pymavlink mavschema.xsd, and only after changing mavschema.xsd do I create PR for mavlink with WIP tags.

hamishwillee, 2021-11-01:
We're trying to phase out WIP such that things will instead appear in development.xml and then appear in common.xml only when they are already "final".

I would just say "no" except that the infrastructure is not playing nice with imported enums. What should happen is that if you declare a duplicate enum then the results should be merged. However this might not happen properly, and in any case there are a couple of exceptions where we've left WIP things in common and it might take a while to get them out.

I'm leaning with not adding WIP to enums and just ignoring this inconsistency. @amilcarlucas What do you think?

python36, 2021-11-02:
> We're trying to phase out WIP
Very nice. For interpreted embedded languages, such unnecessary messages and enumerations load the system.
I was making a generator for CircuitPython and faced the problem of a lot of unnecessary data.

What should dialect owners do? ardupilot.xml has WIP message OBSTACLE_DISTANCE_3D, should it be moved to development.xml or create ardupilot-development.xml file?

hamishwillee, 2021-11-03:
> What should dialect owners do? ardupilot.xml has WIP message OBSTACLE_DISTANCE_3D, should it be moved to development.xml or create ardupilot-development.xml file?

I hadn't realised they are being used in dialects and I probably should have. Generally they are being misused according to their original intent. I'll raise this in the devcall to discuss.

hamishwillee, 2021-11-03:
Had a bit of a chat around> Probably it is a good idea to add support. This would start with an update to https://github.com/ArduPilot/pymavlink/blob/master/generator/mavschema.xsd

Do you want to create a PR for that @python36 ?

python36, 2021-11-03:
@hamishwillee Yes, I create a PR, but after accepting my previous PR that posted a month ago :(

closed 2021-11-03 by python36

amilcarlucas, 2021-11-03:
I think that fine granular WIP tags are a good idea. And the development.xml is also good, but why choose one when you can have both?

hamishwillee, 2021-11-04:
@python36 Sorry we're slow. Everyone has other priorities. If this was breaking vehicles we'd act faster.

>  but why choose one when you can have both?

@amilcarlucas 

Generally because having both doubles the complexity to little benefit. Note that I am actually happy to keep WIP for people who want to use it, but I am hoping that we can avoid it completely in common.xml "eventually". For now we still need it, and excluding enums and enum values is not helpful. I'd also quite like a date field added to WIP to make it easy to track when they were added so we can do a sweep to remove the older ones that never made it.

The long version ...

WIP does not achieve the goal it was intended for. The idea was that WIP would be applied for a few months during final testing. This would ease final roll out testing and still give us an opportunity to make final changes if needed.

What actually happened is that it has been used very early in the design, often even before prototype. Further, WIP is rarely removed at the end of the process. As a result common ended up being littered with WIP for features that are in some cases is complete, and in others have never been implemented.

As part of the governance we're a lot more strict about what goes in common. Essentially once it appears there it should be already frozen - it must have a solid implementation prototype in one flight stack and commitment from another flight stack to take it.
Using development in the interim allows us to enforce this much better than WIP.

WIP may well still have value in dialects to indicate stuff that is still iterating. IMO though it should have an expiry date that encourages removal at a certain point. For now it is useful for common.xml, because we have no way to indicate enums that have been _extended_ as part of new development. So cleanup is difficult if we decide to revert changes.

python36, 2021-11-04:
@hamishwillee 
> Sorry we're slow. Everyone has other priorities.

I understand, but a month is long.
I am already making 3 Mavlink generators for the languages that are used in our products and have come across many very inconvenient and non-obvious points that need to be fixed.

Maybe I can somehow help: donations or work?

> WIP does not achieve the goal it was intended for.

It is necessary to proceed from the purpose of WIP. WIP informs that the message may be removed or changed in the near future. Why include these messages and enums to the stable library?

I proposal adding a checkbox / flag "Include WIP" to the generator with the default value False. This will stimulate to remove the WIP status.

> I'd also quite like a date field added to WIP

Good idea

hamishwillee, 2021-11-10:
@python36 I am sorry that these inconsistencies are causing you problems.  You generators really need to be able to cope with _inconvenient_ syntax since any syntax that can be allowed will at some point be allowed and break your generators in future. If there is a "one true way" to do something then it needs to be mandated in the XSD.

I don't think you can fund faster work here - consensus just takes time. What you can do is highlight the cost to you of the delay so we understand that this does make a real difference to you and why.

>> It is necessary to proceed from the purpose of WIP. WIP informs that the message may be removed or changed in the near future. 

> Why include these messages and enums to the stable library?

Exactly. My intent was they they would NOT be included in the stable library. Never managed to get that PR accepted - was not a priority: https://github.com/ArduPilot/pymavlink/pull/240/

So now the plan is to eventually get all of these out of common.xml so it is a null issue. Development.xml is by definition WIP.

> I proposal adding a checkbox / flag "Include WIP" to the generator with the default value False. This will stimulate to remove the WIP status.

Yes, a generator could do this. As per PR above original plan was to have a define to warn of their use.

> I'd also quite like a date field added to WIP

XSD updates to support WIP in enum and also a date added in https://github.com/ArduPilot/pymavlink/pull/592

python36, 2021-11-10:
@hamishwillee thanks for communicating with me. I have been working with your product for over 5 years now and I want to improve on some points.

> What you can do is highlight the cost to you of the delay so we understand that this does make a real difference to you and why.
This causes my code to be different now. And the further the more. Because of this, it is very difficult for me to release PR with new generators.

https://github.com/ArduPilot/pymavlink/pull/240 opened 3 years ago. This is an important question that was ignored!

hamishwillee, 2021-11-17:
I don't understand what you mean "causes my my code to be different now". 

But in either case, hopefully https://github.com/ArduPilot/pymavlink/pull/592 does what you need!