mavlink/mavlink #1734: Issues about CRC_EXTRA byte.

Hxianyu2001, 2021-11-01:
I can't really comprehend this sentence , 'when the sender **_CALCULATES_** the checksum for a message it adds the CRC_EXTRA byte onto the end of the data that the checksum is **_CALCULATED OVER_**'.
And I'm wondering how the CRC_EXTRA work . 
Does  'adds the CRC_EXTRA byte onto the end of the data that the checksum is calculated over' mean that the final checksum equals to the previous checksum PLUS CRC_EXTRA byte ? Or that the final checksum equals to the previous checksum followed by the CRC_EXTRA byte.

hamishwillee, 2021-11-01:
The CRC calculation is here: https://mavlink.io/en/guide/serialization.html#crc_extra
Perhaps though the best way to understand this is to look at the sending code here: https://github.com/mavlink/c_library_v2/blob/master/mavlink_helpers.h#L256

My _understanding_ of that code is that the CRC calculator is an accumulator . So the final checksum is not equal to the  previous checksum followed by the CRC_EXTRA byte. Instead it is the result of the CRC run over the data, and the CRC_EXTRA run through the CRC accumulator. However I could be wrong. @julianoes is a programmer and might be able to interpret that better.

Hxianyu2001, 2021-11-02:
> The CRC calculation is here: https://mavlink.io/en/guide/serialization.html#crc_extra Perhaps though the best way to understand this is to look at the sending code here: [...]

Thank you for your reply .According to your understanding, I think I've understood the CRC_EXTRA byte.

closed 2021-11-02 by hamishwillee