mavlink/mavlink #1743: MAVLink Generator for Arduino  - how to generate a working library?

AndKe, 2021-11-13:
Many are using the very old Arduino library attached in the first post of this thread:
I wish to build a new, updated library for Arduino, for current Ardupilot vehicles.
The result of:
https://discuss.ardupilot.org/t/mavlink-and-arduino-step-by-step/25566
[image]

Is quite far from the library many programs have been based on, and cannot be used as an Arduino library with convenience functions, at least without some work.

Please describe how to generate a mavlink library, preferably a direct replacement of the old one.

hamishwillee, 2021-11-14:
That library is the MAVLink 1 C library for ArduPilot - you can build it using the mavlink generator as shown in your screenshot or using command line. The full the instructions here: https://mavlink.io/en/getting_started/generate_libraries.html#mavgen

In theory you could also use the pre-built libraries, but they appear to have some problems at the moment: https://github.com/mavlink/c_library_v1

You say that the library is quite different? It shouldn't be that different for MAVLink 1, though it might be quite different for MAVLink 2 - because as you know, MAVLink 2 adds support for a much larger message range, message extensions etc. However I don't know what differences you're looking at so I can't really comment.

hamishwillee, 2021-11-14:
PS Note that the libraries are used all over the place. However we don't say "will work out of the box for specific hardware". If they don't work out of the box with a specific toolchain then you can post the errors and we'll try help you.

closed 2021-11-14 by hamishwillee

AndKe, 2021-11-18:
@hamishwillee  I understand.
This old library: https://discuss.ardupilot.org/uploads/default/original/2X/0/064748b43308485aa9bd0d86fb91d74e17ed8c2a.zip
...have a script that seems to beautifully update it - the few(?) files it contains, make the resulting library work perfectly well for Arduino (over many years now).

MAVLink devs could consider adding the necessary arduino-related files itself, and have a nice option to generate "arduino" library (which is basically C, with the extra files.)
I am using STM32, still, it is very easy to use the Arduino environment for fast results.
Thank you.

hamishwillee, 2021-11-24:
@AndKe I don't see any update script in that folder. The additional files are mostly old dialects, not used.

The only significant difference I see is that this old version has mavlink.h at the top level. Now days you'd instead include the version of mavlink.h in the sub folder of your dialect - so for example /common/mavlink.h.

The source also has a larger max payload size in the current version. I suspect that might affect the runtime memory consumption, but you should still be able to build this.

AndKe, 2021-11-24:
@hamishwillee  - I am sorry, wrong attachment.
Here is the correct one (the one I am using)
[GCS_MAVLink-current.zip](https://github.com/mavlink/mavlink/files/7593622/GCS_MAVLink-current.zip)
 Please note the two files in GCS_MAVLink directory(the library root)
 GSS_MAVLink.cpp and .h is needed for Arduino,  then generate.sh script that seems to work well for updating the library.
It would be very nice if this kind of output was an export option in the mavgen (complete Arduino library)

hamishwillee, 2021-12-01:
Yeah, as I said, we're not going to supply any kind of output or scripts customized to a particular hardware.

The generate.sh is just an automation of a standard mavlink 1 C build. You don't need this. The interesting bit is GSS_MAVLink  which sets some defines to reduce the implementation to just one channel and reduces the size of the max payload to match the data. It also provides some convenience functions for pumping data down that channel. You could document these optimisations if you wan't but we wouldn't expert them with MAVLink.

dizcza, 2023-10-20:
> At some point I will document "how you can customize MAVLink to send less data". It will set similar defines.

Any updates on this part?

hamishwillee, 2023-10-27:
No. Nothing progressed on that.