mavlink/mavlink #1746: MAVLink - easy and more ...manual.. packet assembly.

AndKe, 2021-11-18:
One can use a C function that requests data streams (deprecated way):
mavlink_msg_request_data_stream_send(.......)      // this is a nice and simple way to get it done :)

However, there is no such function for the new way of requesting message rates, like :  mavlink_msg_set_message_interval_send()
There is a function mavlink_msg_message_interval_send   - for replies only, but none for "set" ?

How should I assemble and send https://mavlink.io/en/messages/common.html#MAV_CMD_SET_MESSAGE_INTERVAL  ?
I have searched thru some examples, but failed to find a similar packet.
What makes functions like mavling_msg....._send() exist for a given packet, and not for others?

AndKe, 2021-11-22:
Please read this issue as a plea to consider adding some information about when to expect what, and how to use at http://mavlink.io/en/mavgen_c/  - or in the examples.

olliw42, 2021-11-22:
well, one can argue that it quite clearly expressed here http://mavlink.io/en/mavgen_c/#decode_command

"A MAVLink Command encodes a command defined in a MAV_CMD enum value into a COMMAND_INT or COMMAND_LONG message."

what else would there be to say about it ;)

AndKe, 2021-11-22:
@olliw42   I do see your point - it's easy when you have a certain knowledge about it.
I've searched and asked in this thread: https://discuss.ardupilot.org/t/mavlink-and-arduino-step-by-step/25566/248?u=andre-k
and there is still not one example of how to use COMMAND_* in a context.
Maybe you could please post an example there?  the issue of not using deprecated messages will not be less valid with time.

hamishwillee, 2021-11-24:
@AndKe Yes it would be great to have a simple example of this. It isn't one of my priorities.

Part of the problem is that an example is a bit of a pain - you have to implement a protocol with messages going back and forth - see: http://mavlink.io/en/services/command.html

I could perhaps update http://mavlink.io/en/mavgen_c/#decode_command with links to code examples - but would you have found this?

Some code for decoding an be found here: 
- PX4 - https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L422
- ArduPilot: https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L4637

For sending you can look at https://github.com/mavlink/c_library_v2/blob/master/common/mavlink_msg_command_int.h - as OllieW says, you pack the param1 > 7 in the [MAV_CMDs](https://mavlink.io/en/messages/common.html#mav_commands) that you are interested in into the corresponding slots in the message https://mavlink.io/en/messages/common.html#COMMAND_INT