mavlink/mavlink #1753: mavlink_types.h:92: error: ISO C++ prohibits anonymous structs [-Werror=pedantic]

AleksandrBon, 2021-11-26:
On gcc compilation with "-pedantic" flag error:
mavlink_types.h:92: error: ISO C++ prohibits anonymous structs [-Werror=pedantic]
     };
     ^
mavlink_types.h:17: in definition of macro ‘MAVPACKED’
   #define MAVPACKED( __Declaration__ ) __Declaration__ __attribute__((packed))
                                        ^~~~~~~~~~~~~~~
etc.

This problem is originated in that the c++ standard forbids anonymous structs declarations
[](https://stackoverflow.com/questions/2253878/why-does-c-disallow-anonymous-structs)

And even if mavgenerate.py one can choose to generate c++ headers, core files like mavlink_types.h stays same.

hamishwillee, 2021-12-01:
Thanks.

@julianoes This should probably be moved to Pymavlink repo right?

julianoes, 2021-12-01:
@hamishwillee yes it should.

hamishwillee, 2021-12-01:
@AleksandrBon Could you please repost this in https://github.com/ArduPilot/pymavlink/issues - it is a toolchain issue, and the toolchain is managed in a separate repo.

If you don't want to track this, let me know and I can repost for you.

AleksandrBon, 2021-12-02:
> @AleksandrBon Could you please repost this in https://github.com/ArduPilot/pymavlink/issues - it is a toolchain issue, and the toolchain is managed in a separate repo.
> 
> If you don't want to track this, let me know and I can repost for you.

done https://github.com/ArduPilot/pymavlink/issues/602

Thanks. There is also another interesting moments, i.e. when trying to compile with
-Wswitch-default it gives error that code don't have default switch-case if some files.
And what is more interesting
-Wconversion mavlink_sha256.h:217:40: error: conversion to ‘unsigned char’ from ‘uint32_t {aka unsigned int}’ may alter its value [-Werror=conversion]
That's can potentially cause overflow.

auturgy, 2023-05-24:
Closing - has been closed in pymavlink

closed 2023-05-24 by auturgy