mavlink/mavlink #1755: gimbal v2: GIMBAL_DEVICE_FLAGS_NEUTRAL appears to be redundant

dayjaby, 2021-12-01:
`Set to neutral position (horizontal, forward looking, with stabiliziation), takes presedence over all other flags except RETRACT.`

The same information can be conveyed with a GIMBAL_DEVICE_FLAGS_PITCH_LOCK and pitch, yaw = 0.0 degrees. So what is the gain in having GIMBAL_DEVICE_FLAGS_NEUTRAL?

olliw42, 2021-12-20:
neutral, like retract, may not correspond to pitch,yaw = 0, but a user specified set of values
this is how it is implemented e.g. in ArduPilot with the "old" MOUNT gimbal messages, here the user has parameters to set the gimbal orientation for retract and neutral, respectively. Can't speak for PX4.

hamishwillee, 2021-12-22:
@julianoes Assigned to you :-)

julianoes, 2021-12-22:
Good remark @dayjaby. I guess that might indeed be redundant, it just makes the intention clear.

> neutral, like retract, may not correspond to pitch,yaw = 0, but a user specified set of values

That would make sense but I have not seen it in practice. @olliw42 do you know of an example where that is different than 0,0?

olliw42, 2021-12-22:
it's how it is implemented in ArduPilot, and I certainly don't know what any ArduPilot user is doing, so my answer has to be: 
no, I don't know of any example, but this doesn't mean anything ;)

I however do see use cases for it. I know of STorM32 users which use the STorM32's capability to set an offset as normal default, to e.g. have the camera pointing down by 30° or so. STorM32 also has a recenter function which these users use to get back to it, after e.g. having done some pano or whatever. This neutral would be the Mavlink version of this. (and it's actually how I emulate it in STorM32, it calls recenter)

The question I think should not be if it is redundant, since by these arguments it is not, but whether neutral needs to exits as a separate mode at all as in principle it could be achieved by just setting the pitch,yaw as desired ... i.e., the "state" handling would be moved over to the QCS, companion, or whatever is doing the gimbal. My personal feeling is that it isn't hurting to have it, and can be convenient in some use cases. It also was existing in gimbal v1 and removing it would make transition form v1 to v2 potentially less straightforward. 

So, overall, I think it is good as it is.

julianoes, 2021-12-22:
> e.g. have the camera pointing down by 30°

Yes, that's what I was looking for. I would say it's fine then as it is, except that we might note that in the description. Something like "neutral does not mean 0, 0 but can be some angle that is considered the default".

hamishwillee, 2021-12-22:
That sounds sensible. I have proposed a wording in #1758

dayjaby, 2021-12-23:
@hamishwillee Thanks for the clarification in the description.

closed 2021-12-23 by dayjaby