mavlink/mavlink #1756: Possible misleading description in Mavlink Command Protocol

AleksandrBon, 2021-12-05:
I have a question about description of Command Protocol in [https://mavlink.io/en/services/command.html](url)

It states that there is 2 different types of commands command_int and command_long that differ ONLY (as can be understood from description, because opposite not stated) in that command_int has 2 integer fields for parameters 5 and 6. BUT consequent description about implementation of communication sequence talks only about command_long and uses message field called **confirmation**. And indeed command_int message hasn't such field. So next questions appears - how someone should implement this communication sequence in the case of command_int message?
And next question (for me more important) - what type of retransmission strategy is meant in the case of communication sequence for command_long. My guess it is so called **Stop and wait ARQ**, BUT there is a problem that in classical realization of such retransmission protocol responding side (i.e. drone) must increase confirmation filed in case of successful reception, and not the sender side (look e.g. Bertsekas and Gallager "Data networks", or google).

So maybe I confuse myself and there is another meaning in this case and in implementation for commad_int message too.

hamishwillee, 2021-12-07:
@julianoes Can you comment on this?

I looked at QGC and MAVSDK source and both of them appear to do resending without changing the command_long value for confirmation. They also seem to do nothing special for command_int, which has no confirmation field.

My assumption is that command_int was invented after command_long, and perhaps at that point it was decided the feature is not needed.

But we do need to be clear. If confirmation is updated, what should the recipient side do?

I looked at QGC and MAVSDK. They appear to ignore confirmation (i.e. resending is done

hamishwillee, 2022-03-02:
@julianoes Ping - if you have time!

julianoes, 2022-03-06:
I just read https://en.wikipedia.org/wiki/Stop-and-wait_ARQ and I think you're right @AleksandrBon. We are using a poor version of Stop and wait ARQ. Our version is poor as we don't use any mechanism to deal with duplicate commands or duplicate acks.

One place where we had a problem with this approach was when triggering photos using [MAV_CMD_IMAGE_START_CAPTURE](https://mavlink.io/en/messages/common.html#MAV_CMD_IMAGE_START_CAPTURE) because it would trigger two pictures in quick succession and then not be ready to trigger the next one as the camera was flushing the picture to disk, so we added a sequence number for param4.

So, take away for me is that it is a bit flaky as is until we move MAVLink to a transport that does the reliability part for us, e.g. as part of mavlink 3.

AleksandrBon, 2022-03-07:
@julianoes 
> I just read https://en.wikipedia.org/wiki/Stop-and-wait_ARQ and I think you're right @AleksandrBon. We are using a poor version of Stop and wait ARQ. Our version is poor as we don't use any [...]

Ok. And I think in new Mavlink it is better to use something more advanced like DDS with QoS, because simple Stop-And-Wait ARQ is very bandwidth inefficient and not offers to choose QoS.

closed 2022-03-07 by AleksandrBon