mavlink/mavlink #1757: CMake Build Instructions?

Mentors4EDU, 2021-12-19:
Where are the installation and build instructions for CMake?

hamishwillee, 2021-12-22:
Since cmake isn't a dependency or requirement for MAVLink, I have to ask why you need them?

Mentors4EDU, 2021-12-23:
I am confused though how this is compiled because there is a CMakeLists.txt file and a C Make folder but not make file

vooon, 2021-12-23:
What Makefile you're expecting to see? CMake should generate it, so it shouldn't be under git control. Anyway why do you need it?

hamishwillee, 2022-01-04:
Closing. The existence of https://github.com/mavlink/mavlink/tree/master/cmake which does not appear to have a use is confusing. I'll see if anyone can work out what it is for, and if no one knows, delete it.

closed 2022-01-04 by hamishwillee