mavlink/mavlink #1762: MAVLINK_MSG_ID_EKF_STATUS_REPORT_LEN is different for Mavlink1 vs Mavlink2

ardugitter, 2021-12-30:
I think I found a bug.

mavgen.py -o _mav_one --lang C --wire-protocol 1.0 ./message_definitions/v1.0/ardupilotmega.xml

mavgen.py -o _mav_two --lang C --wire-protocol 1.0 ./message_definitions/v1.0/ardupilotmega.xml

Now compare 

./_mav_one/ardupilotmega/mavlink_msg_ekf_status_report.h

with

./_mav_two/ardupilotmega/mavlink_msg_ekf_status_report.h

The mavlink_ekf_status_report_t typedefs should be identical since both messages are <128 message ID, i.e. Mavlink 1. Mavlink 2 is meant to be backwards compatible with Mavlink 1 but in this case it is not.

The effect of this is that a Mavlink 1 system receiving this message will not be able to decode it due to a bad CRC.

Are my thoughts correct? If so, shall I do a pull request or would someone else like to? I'd prefer someone else do it as I am not set up to do this quickly. Thanks. :-)

ardugitter, 2021-12-30:
Actually I see now that these are called "message extensions". Sorry about the post.

https://mavlink.io/en/guide/define_xml_element.html

closed 2021-12-30 by ardugitter

hamishwillee, 2022-01-02:
Yes. They are not included in the mavlink 2 CRC https://mavlink.io/en/guide/serialization.html#crc_extra