mavlink/mavlink #1764: warning: taking address of packed member of ‘__mavlink_message’ may result in an unaligned pointer value

ardugitter, 2021-12-30:
I get a lot of errors such as this when building mavlink 1 headers in C language:

```
mavlink/common/../mavlink_helpers.h: In function ‘uint16_t mavlink_finalize_message_chan(mavlink_message_t*, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)’:
mavlink/common/../mavlink_helpers.h:86:24: warning: taking address of packed member of ‘__mavlink_message’ may result in an unaligned pointer value [-Waddress-of-packed-member]
   86 |  crc_accumulate_buffer(&msg->checksum, _MAV_PAYLOAD(msg), msg->len);
      |                        ^~~~~~~~~~~~~~
mavlink/common/../mavlink_helpers.h:88:28: warning: taking address of packed member of ‘__mavlink_message’ may result in an unaligned pointer value [-Waddress-of-packed-member]
   88 |  crc_accumulate(crc_extra, &msg->checksum);
      |                            ^~~~~~~~~~~~~~
mavlink/common/../mavlink_helpers.h: In function ‘void mavlink_start_checksum(mavlink_message_t*)’:
mavlink/common/../mavlink_helpers.h:200:11: warning: taking address of packed member of ‘__mavlink_message’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  200 |  crc_init(&msg->checksum);
      |           ^~~~~~~~~~~~~~
mavlink/common/../mavlink_helpers.h: In function ‘void mavlink_update_checksum(mavlink_message_t*, uint8_t)’:
mavlink/common/../mavlink_helpers.h:205:20: warning: taking address of packed member of ‘__mavlink_message’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  205 |  crc_accumulate(c, &msg->checksum);
      |                    ^~~~~~~~~~~~~~
In file included from mavlink/common/../protocol.h:85,
                 from mavlink/common/common.h:30,
                 from mavlink/common/mavlink.h:32,
                 from serial_port.h:66,
                 from serial_port.cpp:55:
mavlink/common/../mavlink_helpers.h: In function ‘uint16_t mavlink_finalize_message_chan(mavlink_message_t*, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)’:
mavlink/common/../mavlink_helpers.h:86:24: warning: taking address of packed member of ‘__mavlink_message’ may result in an unaligned pointer value [-Waddress-of-packed-member]
   86 |  crc_accumulate_buffer(&msg->checksum, _MAV_PAYLOAD(msg), msg->len);
      |                        ^~~~~~~~~~~~~~
mavlink/common/../mavlink_helpers.h:88:28: warning: taking address of packed member of ‘__mavlink_message’ may result in an unaligned pointer value [-Waddress-of-packed-member]
   88 |  crc_accumulate(crc_extra, &msg->checksum);
      |                            ^~~~~~~~~~~~~~
mavlink/common/../mavlink_helpers.h: In function ‘void mavlink_start_checksum(mavlink_message_t*)’:
mavlink/common/../mavlink_helpers.h:200:11: warning: taking address of packed member of ‘__mavlink_message’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  200 |  crc_init(&msg->checksum);
      |           ^~~~~~~~~~~~~~
mavlink/common/../mavlink_helpers.h: In function ‘void mavlink_update_checksum(mavlink_message_t*, uint8_t)’:
mavlink/common/../mavlink_helpers.h:205:20: warning: taking address of packed member of ‘__mavlink_message’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  205 |  crc_accumulate(c, &msg->checksum);
      |                    ^~~~~~~~~~~~~~
In file included from mavlink/common/../protocol.h:85,
                 from mavlink/common/common.h:30,
                 from mavlink/common/mavlink.h:32,
                 from udp_port.h:76,
                 from udp_port.cpp:52:
mavlink/common/../mavlink_helpers.h: In function ‘uint16_t mavlink_finalize_message_chan(mavlink_message_t*, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)’:
mavlink/common/../mavlink_helpers.h:86:24: warning: taking address of packed member of ‘__mavlink_message’ may result in an unaligned pointer value [-Waddress-of-packed-member]
   86 |  crc_accumulate_buffer(&msg->checksum, _MAV_PAYLOAD(msg), msg->len);
      |                        ^~~~~~~~~~~~~~
mavlink/common/../mavlink_helpers.h:88:28: warning: taking address of packed member of ‘__mavlink_message’ may result in an unaligned pointer value [-Waddress-of-packed-member]
   88 |  crc_accumulate(crc_extra, &msg->checksum);
      |                            ^~~~~~~~~~~~~~
mavlink/common/../mavlink_helpers.h: In function ‘void mavlink_start_checksum(mavlink_message_t*)’:
mavlink/common/../mavlink_helpers.h:200:11: warning: taking address of packed member of ‘__mavlink_message’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  200 |  crc_init(&msg->checksum);
      |           ^~~~~~~~~~~~~~
mavlink/common/../mavlink_helpers.h: In function ‘void mavlink_update_checksum(mavlink_message_t*, uint8_t)’:
mavlink/common/../mavlink_helpers.h:205:20: warning: taking address of packed member of ‘__mavlink_message’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  205 |  crc_accumulate(c, &msg->checksum);
      |                    ^~~~~~~~~~~~~~
```

I've also noticed recently that I am having problems verifying the CRC of Mavlink 1 messages that can have Mavlink 2 extension fields. Should I be concerned with all these warnings? I cannot easily tell when they first started appearing unfortunately,

Thanks for reading.

hamishwillee, 2022-01-03:
@julianoes Thoughts on this - I'm not sure how worrying the warnings are.

Generally you should try work with MAVLink 2 unless you absolutely have to work with MAVLink1. Extension fields are ignored in the CRC_EXTRA field.

julianoes, 2022-01-04:
I think this problem is there independent of mavlink 1 vs. 2. Generally, as far as I know, we ignore that warning.

See as an example:

Warning disabled in PX4:
https://github.com/PX4/PX4-Autopilot/blob/e835a7c4ea4bd2c8af179e37c561521a29d715d7/src/modules/mavlink/CMakeLists.txt#L70

All warnings disabled for "system" headers in MAVSDK:
https://github.com/mavlink/MAVSDK/blob/5d2a21f38b7f29664fcd64332b33a1d278811287/src/mavsdk/CMakeLists.txt#L28

Warning ignored on file basis for ArduPilot:
https://github.com/ArduPilot/ardupilot/blob/427c08db2674ad4e02a7907a173b75a7f92bf5a8/libraries/GCS_MAVLink/GCS_MAVLink.h#L34

hamishwillee, 2022-01-04:
Thanks @julianoes 

I've recorded the info that warnings can be ignored here, so closing: https://github.com/mavlink/mavlink-devguide/pull/420

closed 2022-01-04 by hamishwillee