mavlink/mavlink #1765: MAVLINK_MESSAGE_CRCS is incorrect in common/common.h

ardugitter, 2021-12-30:
I had a very strange problem with my CRCs for **certain** Mavlink 1 messages being incorrect. After long and deep investigation, it turned out I was including common/common.h before ardupilotmega/ardupilotmega.h (no, the "MAVLINK_COMMON.H MUST NOT BE DIRECTLY USED" message did not appear when compiling because I already had the relevant #define defined as a result of another mavlink header inclusion elsewhere).

By doing this order of header file inclusion, the MAVLINK_MESSAGE_CRCS array that was used in mavlink_helpers.h at first glance looks legitimate, but further inspection shows that certain CRCs are incorrect.

I'd like to request that MAVLINK_MESSAGE_CRCS is removed entirely from common/common.h to prevent confusion in future. Either that, or please could it be corrected with the right values?

Many thanks and I hope this comment helps someone!

hamishwillee, 2022-01-02:
Thanks, but the problem here is that directly including named dialect libraries. You should instead just include the mavlink.h in the desired top level folder, as documented here: https://mavlink.io/en/mavgen_c/#adding-libraries

closed 2022-01-02 by hamishwillee

ardugitter, 2022-01-02:
> Thanks, but the problem here is that directly including named dialect libraries. You should instead just include the mavlink.h in the desired top level folder, as documented here: https://mavlink.io/en/mavgen_c/#adding-libraries

But why is a corrupt MAVLINK_MESSAGE_CRCS array present in the libraries?

If anything this should be the "most compatible" version, not one with numerous broken CRC values in that allows people to trip up.

hamishwillee, 2022-01-03:
@ardugitter We build all the dialects together, with the lower level includes having less messages and hence CRCs than the higher levels. Since C/C++ dooes not allow redeclaration of the various entities, the way this works is that if you include the correct mavlink.h file you get the correct version of the CRC file for that dialect.

We might in theory build just one file for each dialect with everything which would be less confusing for users. But then developers who read the manual and use the libraries as shown there and in the examples rarely have problems with this.

ardugitter, 2022-01-12:
But surely it would be better to make sure MAVLINK_MESSAGE_CRCS is shorter in length when there are fewer valid CRCs in that array?

That way, the runtime code is aware that "these are the valid CRCs I have available so I will warn if I receive message IDs that exceed the length of the MAVLINK_MESSAGE_CRCS array".

hamishwillee, 2022-01-12:
No. As a user of the array you don't care what the length of the array is. If you get a message that doesn't have a matching value in the CRCS then you know that the message is not compatible with your library.

ardugitter, 2022-01-12:
> No. As a user of the array you don't care what the length of the array is. If you get a message that doesn't have a matching value in the CRCS then you know that the message is not compatible with your library.

If you get a bad CRC, the message might be corrupted somewhere between sender and receiver - that's the whole reason for CRCs in most uses! CDs, TCP/IP and particularly radio environments where corruption is possible. Mavlink is often used over radio telemetry dongles.

Why does Mavlink use the term "CRC" if that is not the case?

hamishwillee, 2022-01-12:
The CRC is used exactly as you say, which is why we use the term CRC. 

But in addition to running the CRC over the content of the message we also include the MAVLINK_MESSAGE_CRCS for the message, which encodes the structure. This is referred to as  CRC_EXTRA. So the CRC at the end of the message encodes both a CRC of the data and of the data structure, allowing us to tell if the message is corrupted and if it is compatible.

This is all documented in https://mavlink.io/en/guide/serialization.html

ardugitter, 2022-01-16:
Okay, I read the link and what you say makes sense.

What doesn't make sense is the fact that if I am including common/common. by mistake or intention in my library, if I receive one of the unsupported messages (i.e. CRC_EXTRA[n] will not be correct), the Mavlink C code as designed will not differentiate between a CRC error in the "checksum" and a CRC error because CRC_EXTRA[n] is incorrect.

Off the top of my head, this happens in  mavlink_parse_char(). If a message is not supported you get a CRC error, even though it might not actually be a CRC error, but rather the specific message is unsupported and so CRC_EXTRA[n] is incorrect.

The problem is this lack of differentiation in the Mavlink C library.

hamishwillee, 2022-02-03:
Yes, that will happen - it's a feature. If the CRC doesn't match because either the message is corrupt or you don't have a matching understanding of what the message is. Either way, you can't use the message and it is discarded.

The design is a problem in two cases 
- if the message passes through a mavlink  router, it will check the CRC and drop the message if the check fails. This sux because the router doesn't usually need to understand the message to forward it, but because it must pass the CRC check you need to always ensure that your router has matching messages. A pain.
- When running over a reliable network like TCP you ideally would like to remove the CRC check and only have the CRC extra. 

So it would be good if they were separate, but they are not, and that isn't changing in this version of MAVLink.

ardugitter, 2022-02-03:
Okay, well I hope this thread helps others who may encounter this and be confused at first glance by the nomenclature. Thanks.