mavlink/mavlink #1770: -waddress-of-packed-member

SimonCanelle, 2022-01-05:
When trying to build my project on a 64bit os, I get this error. I only get in on a 64bit os, not on a 32bit one.

all pointers giving errors are pointers to arrays

OS : raspberry pi os bullseye 64bit with PREEMPT-RT

image: 
[image]
text:
`In file included from /var/sync/MAVLink/common/common.h:2563,
                 from /var/sync/MAVLink/common/mavlink.h:32,
                 from /var/sync/Onboard-SDK-3.9/sample/platform/linux/devbrio/droneIncludes.h:28,
                 from /var/sync/Onboard-SDK-3.9/sample/platform/linux/devbrio/droneFunctions.h:4,
                 from /var/sync/Onboard-SDK-3.9/sample/platform/linux/devbrio/DroneControl.h:4,
                 from /var/sync/Onboard-SDK-3.9/sample/platform/linux/devbrio/DroneControl.cpp:1:
/var/sync/MAVLink/common/./mavlink_msg_obstacle_distance.h: In function ‘uint16_t mavlink_msg_obstacle_di                                                                                                                                    stance_encode(uint8_t, uint8_t, mavlink_message_t*, const mavlink_obstacle_distance_t*)’:
/var/sync/MAVLink/common/./mavlink_msg_obstacle_distance.h:173:158: warning: taking address of packed mem                                                                                                                                    ber of ‘__mavlink_obstacle_distance_t’ may result in an unaligned pointer value [-Waddress-of-packed-memb                                                                                                                                    er]
  173 | sg, obstacle_distance->time_usec, obstacle_distance->sensor_type, obstacle_distance->distances, o                                                                                                                                    bstacle_distance->increment, obstacle_distance->min_distance, obstacle_distance->max_distance, obstacle_d                                                                                                                                    istance->increment_f, obstacle_distance->angle_offset, obstacle_distance->frame);
      |                                                                   ~~~~~~~~~~~~~~~~~~~^~~~~~~~~

/var/sync/MAVLink/common/./mavlink_msg_obstacle_distance.h: In function ‘uint16_t mavlink_msg_obstacle_di                                                                                                                                    stance_encode_chan(uint8_t, uint8_t, uint8_t, mavlink_message_t*, const mavlink_obstacle_distance_t*)’:
/var/sync/MAVLink/common/./mavlink_msg_obstacle_distance.h:187:169: warning: taking address of packed mem                                                                                                                                    ber of ‘__mavlink_obstacle_distance_t’ may result in an unaligned pointer value [-Waddress-of-packed-memb                                                                                                                                    er]
  187 | sg, obstacle_distance->time_usec, obstacle_distance->sensor_type, obstacle_distance->distances, o                                                                                                                                    bstacle_distance->increment, obstacle_distance->min_distance, obstacle_distance->max_distance, obstacle_d                                                                                                                                    istance->increment_f, obstacle_distance->angle_offset, obstacle_distance->frame);
      |                                                                   ~~~~~~~~~~~~~~~~~~~^~~~~~~~~
`

hamishwillee, 2022-01-06:
YOu "probably" don't need to worry. See https://mavlink.io/en/mavgen_c/#build-warnings and #1764

closed 2022-01-06 by hamishwillee

peterbarker, 2022-01-07:
On Wed, 5 Jan 2022, Hamish Willee wrote:

> @peterbarker Have you ever seen an issue from these? [ warning: taking address of packed mem ber of ‘__mavlink_...’ may result in an unaligned pointer value [-Waddress-of-packed-memb er]]

I *believe* it's the case that the warnings are bogus in mavlink, as the 
packed structures' members are re-ordered such that all 4-byte values are 
aligned on 4-byte boundaries (by putting them first), all 2-byte values 
come after those, meaning they're also aligned - and then we throw in all 
the 1-byte values.

Generally you can't ignore the error or you get bus-faults and the like on 
ARM.  But the mavlink-field-packing spec makes it not-a-problem.

IIUC :-)

hamishwillee, 2022-01-12:
Thank you very much.