mavlink/mavlink #1796: Re-implement param_union capability bit

auturgy, 2022-02-13:
https://github.com/mavlink/mavlink/pull/1771 needs to be reverted.

mnumanuyar, 2022-10-25:
this issue is progresing in the referenced issue (https://github.com/mavlink/mavlink/pull/1771#issuecomment-1040852321)

hamishwillee, 2022-10-25:
THanks, yes this was all resolved in the end. 

The flags that should be used now are [MAV_PROTOCOL_CAPABILITY_PARAM_ENCODE_BYTEWISE](https://mavlink.io/en/messages/common.html#MAV_PROTOCOL_CAPABILITY_PARAM_ENCODE_BYTEWISE) and [MAV_PROTOCOL_CAPABILITY_PARAM_ENCODE_C_CAST](https://mavlink.io/en/messages/common.html#MAV_PROTOCOL_CAPABILITY_PARAM_ENCODE_C_CAST) - the `MAV_PROTOCOL_CAPABILITY_PARAM_ENCODE_BYTEWISE` replaces the old union flag.

This needed to be changed because PX4 was setting the value to indicate it supported both c-cast and union style use of a float.

closed 2022-10-25 by hamishwillee