mavlink/mavlink #1821: Seq in message header is useless, and it should be denoted as such :)

olliw42, 2022-03-20:
This topic bothered me for quite a while, and it's the LINK_NODE_STATUS message which triggered me to raise it now.

The simple conclusion is:

The seq field in the message header is useless, and the docs should be adapted to better reflect this.

First off, seq was obviously introduced for the purpose of providing a means to detect loss of packets. Well, in fact, I can't actually know the original motivation since I wasn't around at the time, this purpose is however strongly suggested by comments in the docs like
- "It provides methods for detecting packet drops" (https://mavlink.io/en/#key-features)(https://mavlink.io/en/about/faq.html)
- "Why is the sequence number in the MAVLink header needed? ... Having the sequence in the header allows MAVLink to continuously provide feedback about the packet drop rate ..." (https://mavlink.io/en/about/faq.html)
- "Used to detect packet loss." (https://mavlink.io/en/guide/serialization.html#mavlink2_packet_format)
- "Messages received (estimated from counting seq)" and "Messages lost (estimated from counting seq)" (https://mavlink.io/en/messages/common.html#LINK_NODE_STATUS)
- and similar in some more places

Interestingly, https://mavlink.io/en/guide/serialization.html also states that "Components increment value for each message sent" (we will come back to this below).

Now, obviously, for this to work it is required that - if there is no packet loss - that seq is continuous, i.e., continuously increasing with wraps. 

However, exactly this is not guaranteed, and cannot be guaranteed except in the most primitive situation!

The reason: routing. 

Targeted messages will not reach all components (that's the whole purpose of targeting, right!), and therefore the stream which a given component receives from another component is - by construction - not continuous in seq in general.

Ergo, seq cannot serve the purpose. Q.E.D.

---

One now may think about possibilities to resurrect the situation. Some come to mind immediately:

A) One could attempt to make seq continuous for broadcast messages only. For a component it means that it sends out its messages such that seq is counted up only for the broadcast messages; for targeted messages seq would be undefined.

This approach would allow one to at least detect packet loss for those messages. 

However, one can question if it is of much help. The main origin of lost packets (if not the only one) is the link between ground and air, but the fraction of targeted messages as compared to broadcast messages is not insignificant. Especially "important" messages (= not just stream) may likely be targeted.

B) One could attempt to make seq continuous per link. That is, each link output would change the seq of the messages it emits such that it is continuous on its output. For a component nothing would change, it would send all its messages with continuous seq. Routers however would (have to) change the seq on emission.

In principle this would solve the issue, since now it is guaranteed that on each link the seq is continuous if there is no packet loss.

However: This is not really practically feasible, because (1) changing the seq means that the CRC needs to be recalculated which is somewhat costly and - most seriously - because (2) for recalculating the CRC one needs to know the EXTRA_CRC !!!! This would essentially mean that each link node needs to know all message ids and their EXTRA_CRC, and moreover unknown messages could not be communicated over the link (unless one implements other dirty tricks to recover the extra_crc). This is really not what one wants to have.

C) new Mavlink format

We are obviously hitting here yet another fundamental issue with the Mavlink (v1,v2) protocol.

Unfortunately, suggestions of how to overcome its several issues had been - with quite unreasonable arguments I have to say - bluntly rejected in the past. Pl note that the issue described here would be also overcome with the suggestion made in https://github.com/mavlink/mavlink/issues/1347.

D) Repackaging for lossy links. The mavlink messages can be packed together with a link specific mechanism to detect lost mavlink packets (like its own seq number) or repacked in whatever other way which helps the issue.

This however also comes with disadvantages, like that the data stream needs to be parsed additionally to extract the mavlink packets, which adds latency, and so on. But, as far as I can see, this currently seems to be the only somewhat doable & reasonable approach.

---

Conclusion:

- the docs should be adapted to stop make people believe that seq could allow detection of lost packets
- new messages should not be designed which such ideas in mind, specifically LINK_NODE_STATUS is obviously flawed
- if there is an option E) which I failed to see but which would fully overcome the issue, then the docs should be adapted so it is known to folks, like me :)

Cheers,
Olli

dagar, 2022-03-20:
I agree with most of your points, and my personal interest at this time would be to invest in an option C with scalable routing factored in from the beginning rather than short term band-aid solutions.

olliw42, 2022-03-21:
> invest in an option C with scalable routing factored in from the beginning rather than short term band-aid solutions

personally I would be in the same camp, but my impression is that while a "mavlink v3" is lingering around for long and all arguments are laid out there seems to be also significant resistance. 
Whatever will happen, my argument is that at least the docs should not be misleading

(btw, the thing I'm working on, largely as part of the mLRS project, I've tentatively called mavlinkX lol)

hamishwillee, 2022-03-22:
Yes. The approach for detecting packet loss appears to have been designed on the assumption of single GCS/vehicle channel with no routing. Certainly the issue has really only started to be reported in the last year or so, possibly do to increased use of routers on companion computers.

I understand that the sequence number is also used to detect and drop duplicate messages - i.e. messages that have come to a node via multiple channels.

Yes, MAVLink is incredibly adverse to breaking change. It frustrates the hell out of me. We really need "mavlink 3" if only to batch up a whole bunch of fixes to problems like this. 

Yes, docs need to be fixed, at very least to highlight the issue.

I'm not expecting any kind of short term solution, but I'll raise in the dev call too.

olliw42, 2022-03-23:
> Certainly the issue has really only started to be reported in the last year or so

really? 
surprising ...

> I'm not expecting any kind of short term solution

well, the short term solution should be to adapt the docs ... and to not design new/wip messages like LINK_NODE_STATUS ... :)

dagar, 2022-03-23:
> and to not design new/wip messages like LINK_NODE_STATUS ... :)

LINK_NODE_STATUS was my doing, and it was only after I made an attempt to start using it for things like simple congestion control for mavlink UDP that the extent of these problems became clear to me.

hamishwillee, 2022-03-23:
@dagar [LINK_NODE_STATUS](https://mavlink.io/en/messages/common.html#LINK_NODE_STATUS) is marked WIP - can I remove it? [will it break any existing code]

julianoes, 2022-03-24:
Well laid out @olliw42. I also discovered this when debugging an issue "with lost packets" which as it turned out was of course just routing messing with sequence numbers.

I liked B) until I read the caveat, so that's no solution.

I'm all for C), preferably using an existing transport rather than doing it ourselves.

And you asked for E), so I had the idea that each sender keeps track of the sequence number to each of the targets (sysid/compid) and then sends the messages out accordingly but then I realized that it won't work because you would have to send out broadcast messages multiple times (once per component) otherwise the sequence would be wrong. And that would really defeat the purpose of a broadcast message. So nevermind.

hamishwillee, 2022-03-24:
FWIW I'd probably update the docs "with caveats"
- Sequence number can be used to detect duplicate messages that have reached a mavlink component from different paths.
- Sequence number is commonly used for lost packet detection between two nodes on a "dedicated" channel. Note that if there is a router between the two nodes this approach will not work, because messages that are not routed will appear as packet loss.

We keep on running issues with the current format - mixed CRC and CRC extra, fact that you always have to send both, small address space, large empty spaces inside arrays/messages being impossible to excise. This one too. None of them are the "straw that broke the camel's back" but perhaps they should be "cumulatively".

I think maybe we need an RFC to sort this out - i.e. a comprehensive proposal that brings all the problems, possible solutions, mechanisms to reduce the breakage, costs and benefits together. We need real investment from key stakeholders in at least PX4 and ArduPilot on this.

I've added discussion of this particular issue in the dev call. I'm hoping for us to establish
- How important being able to get link status and packet drop actually is
- How we'd resolve it in an ideal world
- How we might resolve it if we were limited to non-breaking options.

olliw42, 2022-03-24:
@hamishwillee 
I think you will also have to do a search through the repo as seq based comments are spread around. Some links I have given in the top, but they may not be all.

Also, note that this comment "Components increment value for each message sent" is hidden quite "deeply" in the docs, but is very **_crucial_**, with profound consequences. It e.g. means that anyone who wants to use seq for whatever purpose needs to build up and keep a list of all components seen and store the seq per component in this list. For the "detect duplicate messages" purpose it is probably not relevant (I have not thought through all edge cases), but obviously for the 2nd use case. If one wants to continue to give seq some purpose in the docs, this should IMHO be clearly stated too. (it seems to be also quite overlooked)

> large empty spaces inside arrays/messages being impossible to excise

"solving" this feature would be quite difficult to do both in a general way and while still keeping a compact header.  So, I myself am not so much concerned about this.

In the "mavlink X" thing I was mentioning I am working on solving that by a compression algorithm.

Generally, concerning mavlink v3, I guess I would not be as harsh as you :). The one major mistake which had been made is that the protocol pieces (header) and data pieces were not disentangled, but with a proposal like or similar to https://github.com/mavlink/mavlink/issues/1347 it all could be solved. This proposal could be improved by saying that seq is not per component but per link, and that the final crc is not calculated with the header included but only over the payload. This would really do it. 
Such an approach would not be so terribly difficult to implement, since it only needs modest changes to the parsers, and it would be less breaking than the transition form mavlink v1 to mavlink v2. I still find it sad that it was so unreasonably turned down. 
This doesn't mean that I wouldn't vote for a somewhat more deeper change, like e.g. a two-bytes header. I however wanted to bring across that it's all not sooo terrible, IMHO :) To me it seems that it's rather the general admittance than the practical, which is difficult :)

olliw42, 2022-03-24:
> @dagar [LINK_NODE_STATUS](https://mavlink.io/en/messages/common.html#LINK_NODE_STATUS) is marked WIP - can I remove it? [will it break any existing code]

I'm actually planing to come up with a proposal for a message which I tentatively called LINK_NODE_FLOW_CONTROL, which would pair with a message like LINK_NODE_STATUS (I may not raise this proposal this and next week though, still need a bit of time for it LOL). I guess, in view of this, I should then also make a suggestion for a LINK_NODE_STATUS :)