mavlink/mavlink #1829: GLOBAL_POSITION_INT Altitude MSL reference

ThomasDebrunner, 2022-04-04:
In the GLOBAL_POSITION_INT message, the MSL reference geoid is not specified. I think so far this is also not specified and is likely just whatever the GPS receiver outputs. So mostly EGM96? 
That should be standardized, such that when multiple vehicle report their altitude, the reported altitudes are comparable

hamishwillee, 2022-04-05:
Thanks @ThomasDebrunner 

A lot of this stuff was designed on the assumption of single-vehicle systems. I guess you're looking at this in the context of swarms where you need to be sure you have a common understanding of altitude across the network?

Is this a real problem from empirical testing or a theoretical problem from observing the messages? Note that if it is real then I am not sure of the solution - the message doesn't lock that down and is kind of explicitly "whatever your GPS module outputs". I think we'd have to add another field that specifies the WGS84 altitude for this use case.

@auturgy @julianoes Thoughts?

Note, this is also relevant for  [GLOBAL_POSITION_INT_COV](https://mavlink.io/en/messages/common.html#GLOBAL_POSITION_INT_COV), [HIL_GPS](https://mavlink.io/en/messages/common.html#HIL_GPS), possibly others.

ThomasDebrunner, 2022-04-06:
Exactly, the problem arises as soon as you have multi-vehicle setups, since the reported MSL altitudes are not comparable. The GPS_RAW_INT message actually has a (WGS84?) alt_ellipsoid field.

The problem with just using WGS84 is that it is usually very far from the true MSL, so while comparable between vehicles, it is not really interpretable as an altitude above sea level directly.

In case we go with WGS84, the GCS would then have to correct for true MSL for all connected vehicles. Alternatively, we specify a reference, e.g. EGM96 for the MSL altitude field. This may get some pushback from people that prefer (or have GPS that natively support) EGM2008 or newer

auturgy, 2022-04-06:
I'd suggest specifying a default (EGM96 or WGS84 - need to confirm what is currently the assumed behaviour) then have an enum as an optional extension that contains alternates.  I think that would alleviate concerns whilst not be too breaking.

hamishwillee, 2022-04-07:
Thanks @auturgy @ThomasDebrunner 

What I think you might be proposing some kind of `alt_offset` as shown. Is this about right?

name | type | unit | description
-- | -- | -- | --
alt | int32_t | mm | Altitude (MSL). Note that virtually all GPS modules provide both WGS84 and MSL.
relative_alt | int32_t | mm | Altitude above ground
`alt_offset` | int32_t | mm | Altitude offset of field `alt` to the geoid defined by enum geoid. Note that WGS84 is preferred. This can be used to determine 
`geoid_enum` | uint16_t | enum | 0: not supported, 1: WGS84, 2: EGM96 

In that case I think this is a bit complex (i.e. I don't know how to use the offset :-)). 
Instead how about just supply the data for whatever the GPS preferred geoid is?

name | type | unit | description
-- | -- | -- | --
alt | int32_t | mm | Altitude (MSL). Note that virtually all GPS modules provide both WGS84 and MSL.
relative_alt | int32_t | mm | Altitude above ground
`geoid_alt` | int32_t | mm | Altitude relative to geoid_enum geoid. Note that WGS84 is preferred.
`geoid_enum` | uint16_t | enum | 0: not supported, 1: WGS84, 2: EGM96 

In summary, a GPS outputs any geoid-relative value it supports and tells us what it is. 
- GPS has this information, so should be easy to do
- Clients will have to convert to some common reference themselves if needed. 
  - This will be easy for a GCS and hard for a vehicle.
  - I've indicated a preferred format (WGS84) but I'm not sure it matters - if any are different then all clients will have to do the conversion to common reference.
  - We might also use a message to tell the GPS our preferred format but we have the same problem - if any GCS does not support this format, all clients will need to do the reference conversion.
- A system that has not been updated to new message will not get this field.
- A system that has been updated will get 0 for new values if the sending system has not been updated.
- It would be desirable to merge this into relative_alt and just have `geoid_enum`, but I don't see any way to make a non-updated system understand that the `relative_alt` is not relative to ground.
- We might also tell the GCS our preferred format using another message. Aagain 

IFF I thought it would be possible I would simply define a new field `geoid_alt` that must be EGM96 (or other agreed format). That would give a common reference for all new GPS that conform. But that would be hard for GPS makers and I don't think they would do it. 

> The GPS_RAW_INT message actually has a (WGS84?) alt_ellipsoid field.

Yes, but since it doesn't say which format the ellipsoid is in, it is useless for this purpose.

julianoes, 2022-04-07:
I agree it would make sense to properly define it. I never know what is supposed to be what, or what it actually is.

hamishwillee, 2022-04-13:
Might be good if @ThomasDebrunner could come to the mavlink call to discuss - in 30 or so mins :-0

I had a bit of a chat with @auturgy about this. It still isn't completely clear whether two GPS units that both support WGS84 (or whatever) will output the same value for MSL. If they do, then possibly we could just add a field to indicate the unit used for calculating MSL.

I worry this is not the case - in particular a vehicle that supports multiple formats, such as WGS84 and something else, might use an unpredictable format for MSL.

The safest approach is to extending with new fields `geoid_alt` and `geoid_enum`.
(or a new message, though I don't particularly see the value in this case).

hamishwillee, 2022-04-13:
MAvlink call discussion:
1. PX4 with Ublox dumps WGS84 ellipsoid into GPS_RAW alt and the "MSL" value into GLOBAL_POSITION_INT alt.
1. A single consistent ellipsoid reference is needed to meet this use case. Pretty much all systems provide WGS84 so that could be used
   - GCS could convert that to whatever geoid it liked or consistent MSL 
 1. The existing alt is documented as MSL. This is a different value for different modules. We could standardize it as WGS84 but this would likely cause a lot of churn.
 1. We are not sure this is a "real" problem. The issue is for swarms and multi-vehicle network control. In most swarms you use the same vehicle/GPS so you have a common MSL value.
 
To progress this we suggest creating a PR with a new extension field `wgs84_ellipsoid_alt` that categorically dumps that value. This can be used as absolutely consistent reference if supplied and required.

hamishwillee, 2022-04-27:
@ThomasDebrunner Closing this issue as we have a suggestion for it to be progressed (you creating a PR as in ^^^^). Any plan for that?

I won't leave this open because we are unlikely to progress this without you driving this unless something happens to escalate this as a serious problem requiring urgent action.

closed 2022-04-27 by hamishwillee