mavlink/mavlink #1839: Clarification: Home Position Q

flybrianfly, 2022-05-09:
I'm writing code for my [MAVLink library](https://github.com/bolderflight/mavlink) for my [company's flight stack](https://github.com/bolderflight/spaaro) to send the home position when the vehicle is armed. This is actually to facilitate a companion path planning computer to know the origin of the NED frame to convert mission items from a global to a local NED frame.

The [home position message](http://mavlink.io/en/messages/common.html#HOME_POSITION) specifies *q* as a normal vector perpendicular to the surface and aligned with the heading. This makes sense and I assume that q is a quaternion. Is there more information or an example for how q should be computed? Of course quaternions can come in multiple orders (i.e. rotation first, rotation last, etc), so I think a little more clarity would help with general applicability of this message across flight stacks.

Also the *approach_x*, *approach_y*, and *approach_z* variables are a little confusing and seem to imply that for a fixed-wing aircraft, takeoff and landing should happen in opposite directions? Which doesn't make sense to me.

hamishwillee, 2022-05-11:
Hi @flybrianfly 

Yes, q is a quaternion, but I don't know how it should work "for certain".

[HOME_POSITION](http://mavlink.io/en/messages/common.html#HOME_POSITION) used to be set [SET_HOME_POSITION](https://mavlink.io/en/messages/common.html#SET_HOME_POSITION) which would allow you to set `q` directly. That is deprecated but is now set using [MAV_CMD_DO_SET_HOME](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_HOME) which does not have the quaternion. Further, in some flight stacks the home position is stored in a slot for the first mission item as a waypoint, and that does not have a quaternion. Upshot, possibly you cannot set it _directly_.

The implementation in PX4 is here: https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/mavlink/streams/HOME_POSITION.hpp

```cpp
matrix::Quatf q(matrix::Eulerf(0.f, 0.f, home.yaw));
q.copyTo(msg.q);

msg.approach_x = 0.f;
msg.approach_y = 0.f;
msg.approach_z = 0.f;
```

ArduPilot sets it to [1,0,0,0] here: https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L2495
```
const float q[4] = {1.0f, 0.0f, 0.0f, 0.0f};
```
I don't understand quaternions. To me that means the value is unreliable and should probably be deprecated. I'll see if I can get a better answer.

@auturgy @julianoes If you have any thoughts on this, I'd appreciate them!

hamishwillee, 2022-05-11:
Discussed in the MAV call. First shot at addressing it: https://github.com/mavlink/mavlink/pull/1843

PR #1843 (2023-05-25): HOME_POSITION - quaternion can have an invalid value
[HOME_POSITION](http://mavlink.io/en/messages/common.html#HOME_POSITION)  includes a quaternion that specifies the slope and heading of the vehicle on takeoff. 

Both PX4 and ArduPilot provide false values.
- PX4 sets roll and pitch as (level vehicle) but specifies the yaw.
- ArduPilot sets it to [1,0,0,0] (level vehicle facing North).

This change states that if the values are not known to be true then they should be set to `NaN`. I assume this is reasonable, but may not be mathematically. If it is OK , then we should raise bugs against the [... cut]
closed 2023-05-25 by hamishwillee