mavlink/mavlink #1840: Mission Microservice Feature Request: Notify ground station of change to mission items

flybrianfly, 2022-05-10:
I'm developing a path planning / obstacle avoidance companion computer that will live on the vehicle with the flight control system. The ground station sends the flight plan and fence information to the vehicle. The path planning computer re-works the plan to avoid obstacles, terrain, and other UAS operating in the area (using the UTM microservice to receive traffic data). In an ideal world, the path planning computer could send the ground station updated flight plan waypoints to keep the operator aware of what the vehicle is doing and where it is planning to fly. To the best of my knowledge, the [Mission Microservice](http://mavlink.io/en/services/mission.html) does not currently have a mechanism for the vehicle to inform the ground station that the flight plan has changed and to re-download the flight plan from the vehicle. As more autonomy moves to the vehicle, I think this feature would be a benefit to keep the ground station and vehicle in sync.

flybrianfly, 2022-05-10:
Just to add a little more information. I think it would be beneficial to be able to:
1. Tell the ground station to send the vehicle a MISSION_REQUEST_LIST to be able to get the full mission downloaded and updated. This would be useful for when the vehicle has modified or generated many mission items.
2. Tell the ground station to insert or modify a single MISSION_ITEM_INT. This would be useful for quick, on-the-fly adjustments to the flight plan, such as for traffic avoidance.

hamishwillee, 2022-05-11:
@flybrianfly These use cases are actually already discussed in MAVLink.

You can already download the current mission from the vehicle using https://mavlink.io/en/services/mission.html#download_mission.

Where there is a problem is that requires that you download the whole mission, even though it might not actually have changed. Two approaches have both been discussed for this, neither of which have implementations:
- [MISSION_CHECKSUM](https://mavlink.io/en/messages/development.html#MISSION_CHECKSUM) - a message that gets emitted when anything on the plan changes - and you can query for it using MAV_CMD_REQUEST_MESSAGE to find out the checksum.
- [MISSION_CHANGED](https://mavlink.io/en/messages/development.html#MISSION_CHANGED) - support for notification of a partial mission change. Partial missions updates are supported by ArduPilot but not by PX4. Historically they were unsafe (non-atomic) though I understand that may have been fixed now.

With respect to path avoidance in missions, there is an API that does this using vision systems: http://docs.px4.io/master/en/computer_vision/obstacle_avoidance.html#mission-mode-avoidance

YOu might hijack this for things like UTM avoidance. See http://docs.px4.io/master/en/computer_vision/path_planning_interface.html for a bit more info on how it works on PX4. 

ArduPilot does not use this interface, though they do have their own approaches for avoidance.

hamishwillee, 2022-05-11:
Setting closed, as I think those APIs all do what MAVLink requires. I would love to see [MISSION_CHECKSUM](https://mavlink.io/en/messages/development.html#MISSION_CHECKSUM) implemented.

closed 2022-05-11 by hamishwillee

flybrianfly, 2022-05-11:
Excellent, thanks, this appears to fulfill my need. I'll try it out soon.

flybrianfly, 2022-05-26:
FYI, I've tried emitting MISSION_CHANGED, but it's not triggering the ground station to request the updated mission items from the flight computer, unless I'm doing something incorrect. This was tested with QGC v4.1.5 and MP v1.3.77. Understand that's not a fault with MAVLink but rather something that would need to be implemented by QGC and MP.

hamishwillee, 2022-05-26:
> Two approaches have both been discussed for this, neither of which have implementations:

hamishwillee, 2022-05-26:
Yes. :-). I was hoping you might do an implementation!

flybrianfly, 2022-05-26:
I did an implementation for the flight stack side for MISSION_CHANGED, since that seemed like the easiest to quickly put together. An implementation for MISSION_CHECKSUM is on my todo.

hamishwillee, 2022-05-26:
Awesome. I actually was thinking we might do one or the other, and to my mind MISSION_CHECKSUM  is the more useful.

flybrianfly, 2022-05-26:
> Awesome. I actually was thinking we might do one or the other, and to my mind MISSION_CHECKSUM is the more useful.

Yes, I agree completely that MISSION_CHECKSUM is the more useful one, especially if multiple components store the mission items in non-volatile memory, it's a quick check on boot to see that everyone has the correct mission items.

hamishwillee, 2022-06-01:
@flybrianfly IF you do implement MISSION_CHECKSUM on PX4 you should sync up with @peterbarker of ArduPilot fame, as he's looking at implementing this in https://github.com/ArduPilot/ardupilot/pull/20834

Specifically, if as soon as we have an implementation on either QGC or MAVSDK we can validate that the PX4 and ArduPilot implementations of this are compatible.