mavlink/mavlink #1853: Error in generated C++ headers

fisian, 2022-06-02:
I generate headers for C++ with the command `mavgen.py --lang=C++11 --wire-protocol=2.0 --output=include/v2.0 message_definitions/v1.0/all.xml`.

However, there is a problem in the header `common/common.hpp`.
For the enum `MAV_FTP_ERR` for which a member `EOF = 6` is generated.
This causes errors when including the header because `EOF` is a macro defined in cstdio / stdio.h.

I wasn't sure if this should be an issue here or in the pymavlink repo.

hamishwillee, 2022-06-08:
Thanks @fisian - it's an issue here, at least initially. This is what I get generated:

.hpp
```cpp
/** @brief MAV FTP error codes (https://mavlink.io/en/services/ftp.html) */
enum class MAV_FTP_ERR
{
    NONE=0, /* None: No error | */
    FAIL=1, /* Fail: Unknown failure | */
    FAILERRNO=2, /* FailErrno: Command failed, Err number sent back in PayloadHeader.data[1].
		This is a file-system error number understood by the server operating system. | */
    INVALIDDATASIZE=3, /* InvalidDataSize: Payload size is invalid | */
    INVALIDSESSION=4, /* InvalidSession: Session is not currently open | */
    NOSESSIONSAVAILABLE=5, /* NoSessionsAvailable: All available sessions are already in use | */
    EOF=6, /* EOF: Offset past end of file for ListDirectory and ReadFile commands | */
    UNKNOWNCOMMAND=7, /* UnknownCommand: Unknown command / opcode | */
    FILEEXISTS=8, /* FileExists: File/directory already exists | */
    FILEPROTECTED=9, /* FileProtected: File/directory is write protected | */
    FILENOTFOUND=10, /* FileNotFound: File/directory not found | */
};

//! MAV_FTP_ERR ENUM_END
constexpr auto MAV_FTP_ERR_ENUM_END = 11;
```

.h
```c
/** @brief MAV FTP error codes (https://mavlink.io/en/services/ftp.html) */
#ifndef HAVE_ENUM_MAV_FTP_ERR
#define HAVE_ENUM_MAV_FTP_ERR
typedef enum MAV_FTP_ERR
{
   MAV_FTP_ERR_NONE=0, /* None: No error | */
   MAV_FTP_ERR_FAIL=1, /* Fail: Unknown failure | */
   MAV_FTP_ERR_FAILERRNO=2, /* FailErrno: Command failed, Err number sent back in PayloadHeader.data[1].
		This is a file-system error number understood by the server operating system. | */
   MAV_FTP_ERR_INVALIDDATASIZE=3, /* InvalidDataSize: Payload size is invalid | */
   MAV_FTP_ERR_INVALIDSESSION=4, /* InvalidSession: Session is not currently open | */
   MAV_FTP_ERR_NOSESSIONSAVAILABLE=5, /* NoSessionsAvailable: All available sessions are already in use | */
   MAV_FTP_ERR_EOF=6, /* EOF: Offset past end of file for ListDirectory and ReadFile commands | */
   MAV_FTP_ERR_UNKNOWNCOMMAND=7, /* UnknownCommand: Unknown command / opcode | */
   MAV_FTP_ERR_FILEEXISTS=8, /* FileExists: File/directory already exists | */
   MAV_FTP_ERR_FILEPROTECTED=9, /* FileProtected: File/directory is write protected | */
   MAV_FTP_ERR_FILENOTFOUND=10, /* FileNotFound: File/directory not found | */
   MAV_FTP_ERR_ENUM_END=11, /*  | */
} MAV_FTP_ERR;
#endif
```

@julianoes Thoughts? Solutions?

julianoes, 2022-06-09:
The `enum class` is namespaced correctly, however, the string EOF is presumably still already defined by a macro.

Have you tried including `<cstdio>` instead of `stdio.h`?

hamishwillee, 2022-06-15:
@fisian Ping re comment above ^^^

fisian, 2022-06-15:
I don't include `<cstdio>` or `stdio.h` myself. The error occurs even when I try to compile a file that only includes the generated header.

I also don't know from where exactly the EOF macro is included or where it is defined.
From a quick investigation it seems that it is indirectly included by multiple headers of the standard library (of which the first occurence in the generated file is `<sstream>`).

danx0r, 2022-09-19:
I have this error - extra */  in 2 places in vehicle_command__struct.h - presume C++ compiler was more forgiving previously

hamishwillee, 2022-09-21:
@julianoes Mean anything to you?

julianoes, 2022-09-22:
Can someone create a minimal example to reproduce? That would help me understand how it all comes together, thanks.

fisian, 2022-09-23:
I don't know what the error is @danx0r describes. It's not the issue I reported and I don't have anything called vehicle_command. A quick search suggests it is specific to the PX4 firmware.

spectre-solutions, 2022-12-19:
Hi all. The problem which @fisian  described still exists. 
I propose a solution here: https://github.com/ArduPilot/pymavlink/pull/761
Should I create an Issue for pymavlink? 
Thanks

hamishwillee, 2022-12-21:
@spectre-solutions Thanks for that - https://github.com/ArduPilot/pymavlink/pull/761 is the right place to post this. It may take a while for the team to look at it though.

spectre-solutions, 2023-01-03:
Issue was fixed in pymavlink https://github.com/ArduPilot/pymavlink/issues/762
Are there some rules about updating pymavlink submodule? We would be thankful if someone could do this.

hamishwillee, 2023-01-04:
@spectre-solutions No rules other than basic sanity check. Anyone can create a PR to update - which I have done and merged in https://github.com/mavlink/mavlink/pull/1940

hamishwillee, 2023-01-04:
Thanks for fixing this up!

closed 2023-01-04 by hamishwillee

spectre-solutions, 2023-01-04:
Great, thank you as well!

PR #1940 (2023-01-04): Update pymavlink
Fixes #1853