mavlink/mavlink #1862: Wireshark showing 0 for negative int32

ThomasDebrunner, 2022-07-04:
With mission items with negative y (negative int32 value), the wireshark plugin incorrectly shows 0, even though the data is clearly not 0 (as seen in the hex dump)

[image]

hamishwillee, 2022-07-06:
@peterbarker Can you look at this please. I cross posted to the generator repo https://github.com/ArduPilot/pymavlink/issues/701

hamishwillee, 2022-07-07:
@ThomasDebrunner Request for a test file here: https://github.com/ArduPilot/pymavlink/issues/701#issuecomment-1178335416

hamishwillee, 2023-02-15:
@ThomasDebrunner - if you want this looked at we need test file please.

ksols, 2026-10-04:
Same root cause as #2073. It was a Wireshark Lua bug (signed int32 values converted as unsigned, giving 0 on Mac), fixed in Wireshark 4.2.5 and worked around in ArduPilot/pymavlink@052a201a. Reproduced and verified in https://github.com/mavlink/mavlink/issues/2073#issuecomment-5984339482. I think this can be closed as a duplicate.

closed 2026-10-07 by hamishwillee