mavlink/mavlink #1874: pymavlink sysid problem ?

patatessever, 2022-08-09:
Hi. I am using pymavlink to communicate between 3 devices using a hotspot over  a UDP port. I have 1 GCS code and 2 Drone codes. I am just trying to recieve and sent heartbeat. The code of GCS is :
`
     from pymavlink import mavutil

    the_connection = mavutil.mavlink_connection('udpin:localhost:14540')
    while True:
        the_connection.mav.heartbeat_send(mavutil.mavlink.MAV_TYPE_QUADROTOR,
                                                    mavutil.mavlink.MAV_AUTOPILOT_GENERIC, 0, 0, 0)

        the_connection.wait_heartbeat()
        print("Heartbeat from system (system %u component %u)" % (the_connection.target_system, 
        the_connection.target_component))
`
and the code inside the drones is 

`

    from pymavlink import mavutil

    the_connection = mavutil.mavlink_connection('udpout:10.42.0.1:14540')

    while True:

        the_connection.mav.heartbeat_send(mavutil.mavlink.MAV_TYPE_GCS,
                                                    mavutil.mavlink.MAV_AUTOPILOT_INVALID, 0, 0, 0)

        the_connection.wait_heartbeat()
        print("Heartbeat from system (system %u component %u)" % (the_connection.target_system, 
        the_connection.target_component))`

I am able to recieve and transmit heartbeat yet the system id thing is not really working. The received system id solely depends on the MAV_TYPE which we are definig this on GCS. I dont really understand how to give specific ID for systems and how the GCS can see drones with different id's ?

hamishwillee, 2022-08-09:
This repo is for mavlink "issues/bugs".

That said, I think the problem here is that you are not _setting_ any ids so you are getting the default, which is usually `1`

I THINK you do this like (say). You'd set different IDs for each node.

```
the_connection= mavutil.mavlink_connection('udpout: your address', source_system=125)
```

If that doesn't work, I suggest you ask on the Pymavlink gitter channel: https://gitter.im/ArduPilot/pymavlink

hamishwillee, 2022-08-09:
PS This is documented in the last line of the section https://mavlink.io/en/mavgen_python/#setting_up_connection

closed 2022-08-09 by hamishwillee