mavlink/mavlink #1884: Clarify how bitmasks should work with arrays

Erk-, 2022-09-02:
In pr #1322 the enum `ESC_FAILURE_FLAGS` was added.

It is defined to have the type [`uint16_t[4]`](https://github.com/mavlink/mavlink/blob/92d2a58df22955abd92ab06f3a67a21ee7a5fa85/message_definitions/v1.0/common.xml#L6986)

It is to me not clear how the values should be handled when they go above `1<<15`.
Should it for all intents and purposes be seen as a `uint64_t` for that should be given?

As far as I can tell this is the only place where a array is used for a bitmask so maybe this should be changed into a `uint64_t` or similar.

Erk-, 2022-09-02:
Further reading of the pr makes it seem that it should be read as 4 different bitmasks of the format described by `ESC_FAILURE_FLAGS` is this the correct way of looking at it?

hamishwillee, 2022-09-03:
Yes. Each entry in the array represents a different ESC.

Erk-, 2022-09-05:
Would something like the following not make more sense?
At least this removes a special case I would have to handle for code generation
```patch
-      <field type="uint16_t[4]" name="failure_flags" enum="ESC_FAILURE_FLAGS" display="bitmask">Bitmap of ESC failure flags.</field>
+      <field type="uint16_t" name="failure_flag_0" enum="ESC_FAILURE_FLAGS" display="bitmask">Bitmap of ESC failure flag for index 0.</field>
+      <field type="uint16_t" name="failure_flag_1" enum="ESC_FAILURE_FLAGS" display="bitmask">Bitmap of ESC failure flag for index 1.</field>
+      <field type="uint16_t" name="failure_flag_2" enum="ESC_FAILURE_FLAGS" display="bitmask">Bitmap of ESC failure flag for index 2.</field>
+      <field type="uint16_t" name="failure_flag_3" enum="ESC_FAILURE_FLAGS" display="bitmask">Bitmap of ESC failure flag for index 3.</field>

```

hamishwillee, 2022-09-06:
@Erk- This message is long time deployed and we can't change it without breaking stuff.

closed 2022-09-06 by hamishwillee