mavlink/mavlink #1907: Dart support?

penzi520, 2022-10-27:
can support Dart?

hamishwillee, 2022-10-27:
Dart is not supported by the project, but there is nothing to stop others writing Dart libraries. Can't comment on how compliant they might be. For example: https://github.com/nus/dart_mavlink

The list of supported languages is here https://mavlink.io/en/#mavlink-project-generatorslanguages

closed 2022-10-27 by hamishwillee

LiangHuangBC, 2022-10-27:
Is it possible to build a controller like QGC by flutter?  Any helpful repo or link? Thanks a lot. @hamishwillee

hamishwillee, 2022-10-27:
I have no idea. All I would be doing if I sent you a link would be googling. But if I had to start anything like that, I'd look, at MAVSDK.

LiangHuangBC, 2022-10-28:
Thanks. MAVSDK is fine.

explorer-source, 2024-07-01:
@LiangHuangBC have you made it work? I'm also trying to build custom QGC with flutter