mavlink/mavlink #1942: Code generation bug for mavlink/c_library_v2

Uiopio, 2023-01-18:
In commit c8927a0359a238133a8b76f67348a39094aa2905 dated 11/24/2022, enum MAV_CMD was removed from common/common.h.
Link to the commit:
https://github.com/mavlink/c_library_v2/commit/c8927a0359a238133a8b76f67348a39094aa2905#diff-34c932128256ee886b3a8581b5f11a1c38717aaa9d228189f1ce12e823f32

[image]

OlegErmolaev, 2023-01-18:
Oh damn man, I've the same problem as u, so can't get the actual version of c_library for my project

hamishwillee, 2023-02-01:
This is likely caused by a known bug in Pymavlink. Essentially if you build a library that has enums with duplicate values, they are merged - but only the top level library gets the merged enums.
Since mav_cmds are enum values, this means that if you build ardupilotmega.xml, the generated ardupilot would get those mav commands, and they would be stripped out of common.xml. 

The libraries generated here are all build one after the other. That is supposed to be compatible, but because of the bug, if the order is wrong you might end up with missing commands. It is fragile.

I raised a PR to fix this years ago:  https://github.com/ArduPilot/pymavlink/pull/544 - still not accepted or alternative proposed. PX4 and ArduPilot now avoid this by building the library needed from source. That's the easiest way to do this.

olliw42, 2023-02-03:
@OlegErmolaev 
if you are ok with using a different code base you could consider [fastMavlink](https://github.com/olliw42/fastmavlink), which doesn't have this issue. It also has a [pymavlink-mavgen mimicry](https://github.com/olliw42/fastmavlink#pymavlink-mavgen-mimicry) mode which should ease porting.

hamishwillee, 2026-08-25:
Closing. The issue still exists, but there is nothing I can do about it. Feel free to try push https://github.com/ArduPilot/pymavlink/pull/544

closed 2026-08-25 by hamishwillee