mavlink/mavlink #1946: Convinience function(s?) blindly copy 16bytes, - even when less are provided... resulting in somehow corrupt packets.

AndKe, 2023-01-31:
mavlink_msg_param_value_pack 
uses:
mav_put_char_array(buf, 8, param_id, 16);
mav_array_memcpy(packet.param_id, param_id, sizeof(char)*16);

The compiler does not like this, and it compiles ... but produced odd packets when the parameter name comes from a struct. - oddly enough this does not happen if the parameter is just inside a const char.

When the incoming array with parameter name is for example 5 bytes ... the rest of the copied data is random.
Actually -- this produces an ok first param packet, but the consecutive packets are no good. confusing MAVProxy.

I have an Arduino application that can reproduce this. - the discussion is here: https://forum.arduino.cc/t/when-a-stuct-char-is-not-a-char-anymore/1083746/29

hamishwillee, 2023-02-01:
Thank you. Can you please repost in https://github.com/ArduPilot/pymavlink as an issue? That's where that particular code is maintained.

@peterbarker FYI, This sounds odd, but I don't know how to debug it. The useful discussion is in https://forum.arduino.cc/t/when-a-stuct-char-is-not-a-char-anymore/1083746/35

peterbarker, 2023-02-01:
Warning is given that you must supply the full-on buffer:
```
/**
 * @brief Send a param_value message
 * @param chan MAVLink channel to send the message
 *
 * @param param_id  Onboard parameter id, terminated by NULL if the length is le
ss than 16 human-readable chars and WITHOUT null termination (NULL) byte if the 
length is exactly 16 chars - applications have to provide 16+1 bytes storage if 
the ID is stored as string
```

AndKe, 2023-02-01:
@peterbarker Yes. I based my code on information from: https://mavlink.io/en/messages/common.html#PARAM_VALUE , then the convenience function.
The warning says:
" terminated by NULL if the length is less than 16 human-readable chars"
therefore I understand it as shorter than 16chars are ok when null-terminated - (which I did).

Then it says: "applications have to provide 16+1 bytes storage if the ID is stored as a string"  
- yes, but I was only sending 8chars , did not expect to store parameters, only known, short parameters - so my char arrays were defined with sufficient size, not 16 chars each.

Please note that my packets had 4 chars and were null-terminated, but random data at the end of the 16chars messed it up.
To me - the solution was to provide a char[16] array with 0x0 in all unused positions after the last character.

An appropriate warning for me would be that the provided char[16] had to be no less than 16chars. (due to copy method) - null-filled after the last char.
Please let me know what you think - I agree that if the warning is adapted - then this issue is resolved.

peterbarker, 2023-02-02:
>  Please let me know what you think - I agree that if the warning is adapted - then this issue is resolved.

Well, the current wording has confused at least one person - so if you want to come up with some better wording in there I'll definitely look at it!

AndKe, 2023-02-02:
I suggest: 

```@param param_id  Onboard parameter id, requires char[16]. If the parameter ID is less than 16 human-readable chars, fill the remaining chars with (NULL) byte(s). Applications have to provide 16+1 bytes storage if the ID is stored as a string.
```

This would prevent any potential issues with random data after NULL, and conveys the other points as well.

hamishwillee, 2023-02-08:
The problem with the original text is that it implies that you can have just one NULL char to terminate the string. In practice the parser probably assumes you will zero-fill to the end so that normal empty-byte truncation can happen on the message.
In any case, I don't think we need to say "_requires char[16]."._ " if you make it clear the it may be multiple bytes that have to be zero-filled

So perhaps ?:

> Onboard parameter id. If the parameter ID is less than 16 human-readable chars, zero-fill the remaining chars in the array (terminate with NULL bytes). 
> Applications have to provide 16+1 bytes storage if the ID is stored as a string.

@peterbarker @AndKe Are you OK with this change?

AndKe, 2023-02-08:
@hamishwillee yes, it's perfectly fine for me- this text would save me a lot of troubleshooting. - and may do so for others.
Please search and replace - searching for "16 human-readable" I have found eight similar texts in common MAVlink , and two in the ardupilotmega dialect.

hamishwillee, 2023-02-15:
FYI still going on this. Zero filling is not the answer.

AndKe, 2023-02-15:
Yes, I totally agree - it would be best to avoid zero-filling - as long as the recipient only reads ASCII until zero - I do not even understand why having random data in the remaining position cocked up the whole thing.
That's why I spent a good while troubleshooting this.  I checked my ASCII bytes, and 0x0 , and ignored the random bytes following.

hamishwillee, 2023-02-15:
I don't understand either - though if you follow the rabbit hole in https://github.com/mavlink/mavlink/pull/1950 it might provide some insights.

hamishwillee, 2023-02-15:
The point is that if you provide a buffer of the size indicated this will work across implementation, and if you don't, you might run into implementation specific issues. For me the question is "where" you provide this information - you ran into this in the PARAM_VALUE but there are lots of messages with fields that null terminate and I'm not sure I want to make this clear "everywhere".

shancock884, 2023-10-15:
This is related to [Pymavlink Issue #725](https://github.com/ArduPilot/pymavlink/issues/725), which suggests to use strncpy instead of memcpy when serializing char[n] fields.
This would then stop copying characters once a null is hit, or N chars is reached (whichever is first), allowing for shorter  (null terminated) strings to be used when the coder feels this is appropriate for them.
And it will fill the rest of the target buffer with nulls for good measure, which is nice for the message-truncation-before-sending thing.

julianoes, 2023-11-22:
Just my two cents: I have seen this implemented for various messages in various places in the past, by me as well as others, definitely a tricky API.

shancock884, 2024-02-14:
@hamishwillee @julianoes ... for info, PR created: https://github.com/ArduPilot/pymavlink/pull/922

closed 2025-06-18 by tridge