mavlink/mavlink #1952: Installation instructions

pjalocha, 2023-02-26:
Would be very good to add simple installation instructions, as it is not trivial and would greatly speedup practical usage.
For example on Linux like Raspian/Ubuntu I had to run the following, before I could use #include <mavlink.h> in my code.

sudo apt-get install python3-pip
pip3 install pymavlink
git clone https://github.com/mavlink/mavlink.git
cd mavlink/
git submodule update --init --recursive
mkdir build
cd build
cmake ..
make
sudo make install

And then, for g++, the following is needed:

-I/usr/local/include/mavlink/v2.0/common/

which decides which MAVlink version one uses.

Once this is known/done all works as expected, but it takes time to find out, thus why not add such to README ?

hamishwillee, 2023-02-26:
These look pretty simple

https://mavlink.io/en/getting_started/installation.html
https://mavlink.io/en/getting_started/generate_libraries.html

Using libraries then depends on which language you're using. So for example, https://mavlink.io/en/mavgen_c/

I guess arguably you might state "C" is the most important language so simple instructions for that setup in the readme wouldn't go astray.

hamishwillee, 2023-03-08:
@pjalocha Looking at your instructions in more detail:
1. You do not need to do `pip3 install pymavlink`. That installs a prebuilt Python library that uses a slightly modified version of common.xml for ArduPilot, and not the official common.xml.
2. I have added a quickstart to the readme which summarizes how you would set this up on Ubuntu 20.04: https://github.com/mavlink/mavlink#quickstart.
   Note that this builds using `pymavlink.tools.mavgen`, thereby fetching the particular submodule of Pymavlink associated with the cloned repo.

Looking at the rest of your instructions what I think you are doing is using cmake to "build" the library and install it system wide. What I think that does is build all the C libraries for common and dialects and install them  to `/usr/local/include/mavlink/v2.0/common/`.
Is that right?
```
mkdir build
cd build
cmake ..
make
sudo make install
```

So then you're saying if you wan to use these in your code, when g++, the following is needed:
   ```
    -I/usr/local/include/mavlink/v2.0/common/
   ```
Is that all correct.

IF so, then I need to think on this a little more. The docs are written with a focus on writing your own dialect and are general to C, Python and so on. What I think your focus might be is using the existing C libraries in your own project. 
So for you the important thing is having a system wide install and a predictable way to access the libraries. 

Does that make sense?

pjalocha, 2023-03-08:
Hello, and thank you.

I am not an expert on doing correct library installations.
What I needed was to complie MAVlink in C/C++ code of Raspberry-class CPU and system, as well as on Ubuntu Linux.
I did not know how to bite this library, nothing seemed to work (mavgenerate.py did not either), after I "git clone" so I began to search around and found this solution: I was able to include and compile mavlink headers which was exactly what I needed.

The next think I will need is to compile MAVlink headers with ESP32-class C/C++
but possibly I can tell the compiler to use the same system-wide installed headers, I don't know yet.

hamishwillee, 2023-03-08:
@julianoes Can you advise here?

In particular, my assumption [here](https://github.com/mavlink/mavlink/issues/1952#issuecomment-1459398182) is that if you build using `cmake` and then do `sudo make install` then what you are doing is generating C libraries to a standard location on ubuntu. 

In the [using C instructions](https://mavlink.io/en/mavgen_c/#adding-libraries) we say you compile like this, pointing to the generated libraries, because our instructions assume you will generate locally.
```
$ gcc ... -I generated/include -I generated/include/common ...
```
What this suggests, and primarily would be if you wanted to just build the libraries for use by anyone, is that you might do
```
$ gcc ... -I /usr/local/include/mavlink/v2.0/common/
```
I assume g++ is called by gcc or is a similar toolchain.

Specifically:
- Is my assumption correct
- Does it make sense to install to a system wide location like this, and under what conditions?
- When would using our specific path-relative instructions make more sense? Obviously if you're iterating your messages a lot then not having to install is a "good thing".

- Should these commands work for all linux platforms - such as ESP32. My thinking is that you don't need to system install or build this way if you don't want - you can build the libraries and copy them across if you want, or just download the pre-built versions, or whatever.
  - The ONLY issues w.r.t to different platforms are things like the size of built library and the endianness. But these all get set by macros when building the thing that uses the library, so should affect none of these instructions.
 About right?

hamishwillee, 2023-04-05:
@julianoes Pinging you again on this. Do you think you can help?

julianoes, 2023-04-10:
I'd rather not document something that looks somewhat odd and potentially broken.

It would be nice to have a simple CMakeLists.txt that works in a cmaky way:

1. Properly install the headers in the default cmake path (`/usr/local/include`, and `/usr/local/lib`).
2. Add the cmake config/target files to the system so MAVLink can be found using `find_package`
3. Add the include directories automatically, so all you need to do is `target_link_libraries(my_exe MAVLink::mavlink)` and that's eventually saying that we depend on library files (in this case none) and includes.

hamishwillee, 2023-04-12:
Thanks, makes sense. So we need to get a proper CMakeList.txt - @junwoo091400 is this something you could help with, or find someone to help us with?

See note ^^^^ for context - https://github.com/mavlink/mavlink/issues/1952#issuecomment-1501340493

julianoes, 2025-09-03:
Fixed in #1977, pretty sure, otherwise make a PR suggestion.

closed 2025-09-03 by julianoes