mavlink/mavlink #1968: Specification of board_version in AUTOPILOT_VERSION

JonasVautherin, 2023-03-31:
The [AUTOPILOT_VERSION](https://mavlink.io/en/messages/common.html#AUTOPILOT_VERSION) message contains a `board_version` field that is specified as:

```
HW / board version (last 8 bits should be silicon ID, if any). The first 16 bits of this field specify https://github.com/PX4/PX4-Bootloader/blob/master/board_types.txt
```

I am not sure how to understand "the first 16 bits". Are they the 16 least significant bits or the 16 most significant ones? Intuitively I would take "first bits" as "left-most" and hence MSB.

How should it be interpreted? And does somebody know how PX4/Ardupilot interprete this?

hamishwillee, 2023-04-05:
Multiple bytes are serialized little-endian, so I assume the last 8 bits are the bits with the lowest address. The definition seems to indicate 32 bits:
```
16: board type | 8: nothing | 8: silicon ID.
```

@peterbarker @bkueng Can you advise how the flight stacks interpret this? Is the 8: still used for nothing?

If someone were to come to this completely new, and want to know how to write to and read from the data, what else would you tell them?

JonasVautherin, 2023-04-05:
I am a bit confused because I have seen code that does this (with PX4):

```c++
        if (msg.msgid == MAVLINK_MSG_ID_AUTOPILOT_VERSION) {
            mavlink_autopilot_version_t autopilot_version;
            mavlink_msg_autopilot_version_decode(&msg, &autopilot_version);

            uint32_t target_number = (autopilot_version.board_version) & 0xFFFF;
            std::string target_name = "";
            if (target_number == 50)
                target_name = "PX4 FMU V5";
            else if (target_number == 51)
                target_name = "PX4 FMU V5X";
            else if (target_number == 52)
                target_name = "PX4 FMU V6";
            else if (target_number == 53)
                target_name = "PX4 FMU V6X";
        }
```

So apart from the fact that the code above (wrongly) assumes that there is only a `board_type` in that int32, my understanding is that if it works, it means that what's received from PX4 looks more like below, right?

```
8: nothing (would be silicon ID) | 8: nothing | 16: board type
```

Which would hint towards saying that the "first 16 bits" of the specs are the lower bits (LSB).

Would that make sense?

hamishwillee, 2023-04-05:
@JonasVautherin I think so - because otherwise target_number would be much bigger. But I'm not a coder, so lets get @julianoes involved.
FYI, ArduPilot code around this is here: https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L2587

JonasVautherin, 2023-04-06:
Oh, thanks for the Ardupilot link! I see this:

```c++
    uint32_t board_version { uint32_t(APJ_BOARD_ID) << 16 };
```

which to me seems like they put the `board_id` on the higher bits (MSB), which would be the opposite of the PX4-related code I pasted above :sweat_smile: :sweat_smile: :sweat_smile:.

hamishwillee, 2023-04-06:
Probably. The other way to check is to grep QGC. If it has different code for both cases you'd be right for sure.

JonasVautherin, 2023-04-06:
I tried, but I can't find `board_version` there :thinking:.

hamishwillee, 2023-04-06:
Hmmm. I would have thought it was somewhere in firmware update. But I'm only hear for my charm and good looks.

patrickelectric, 2024-06-13:
When checking an old pixhawk v2, the bootlader answer as:
```
PX4BoardInfo(board_id=9, bootloader_version=4, flash_size=2080768)
```
That appears to be correct, using QGC approach, we have the same result as board_id=9, and the board_id identification matches what is described [in QGC source code](https://github.com/mavlink/qgroundcontrol/blob/1cd55296d889f3b201e32a321f5293d70526a649/src/VehicleSetup/FirmwareUpgradeController.cc#L55).

But when using mavlink AUTOPILOT_VERSION:
```
{
  "message": {
    "type": "AUTOPILOT_VERSION",
    "board_version": 17,
    "product_id": 17,
    "vendor_id": 9900,
    ...
  },
}
```

As you can see, it appears that there is a difference between the numbers provided by MAVLink and by bootloader, and as you can also check, there is no 17 board identification in QGC source code and also in [PX4 table](https://github.com/PX4/PX4-Bootloader/blob/main/board_types.txt).

As we know, from QGC source code comment:
https://github.com/mavlink/qgroundcontrol/blob/1cd55296d889f3b201e32a321f5293d70526a649/src/VehicleSetup/Bootloader.cc#L137-L142
```cpp
        // Older V2 boards have large flash space but silicon error which prevents it from being used. Bootloader v5 and above
        // will correctly account/report for this. Older bootloaders will not. Newer V2 board which support larger flash space are
        // reported as V3 board id.
        if (_boardID == boardIDPX4FMUV2 && _bootloaderVersion >= _bootloaderVersionV2CorrectFlash && _boardFlashSize > _flashSizeSmall) {
            _boardID = boardIDPX4FMUV3;
        }
```
Weirdly, that is not the case here, as you can see in the bootloader version provided it's a version 4, that should not have this problem.

Well, not sure how to proceed from here, but it appears that the bootloader information is the most correct way to proceed. Any tips about how to check and use MAVLink as a safe source for board identification ?

When using a _Pixhawk 4 (Px4 FMU - V5 default)_, everything matches and we are happy.

hamishwillee, 2024-06-19:
So on PX4, MAVLink gets the value of this field using this function. 

```
uint32_t px4_board_version(void)
{
#if defined(__PX4_NUTTX)
	return CONFIG_CDCACM_PRODUCTID;
#else
	return 1;
#endif
}
```

For FMUv5 we have `CONFIG_CDCACM_PRODUCTID=0x0032` - which is TARGET_HW_PX4_FMU_V5  =  50 in board config.

Dec 17 / 0x0011 is what PX4 is setting for FMUv3. (`CONFIG_CDCACM_PRODUCTID=0x0011`)

So I would say that this is a bug the PX4 source for   `ot\boards\px4\fmu-v3\nuttx-config\nsh\defconfig`.

In which case, isn't this a PX4 error?