mavlink/mavlink #1969: `storm32` and `common` has inconsistent usage for `GIMBAL_DEVICE_CAP_FLAGS`

patrickelectric, 2023-04-01:
From: common
https://github.com/mavlink/mavlink/blob/d6d86d3f0c90a67c275d6833931f7c712b858dba/message_definitions/v1.0/common.xml#L6952

And storm32
https://github.com/mavlink/mavlink/blob/d6d86d3f0c90a67c275d6833931f7c712b858dba/message_definitions/v1.0/storm32.xml#L318

GIMBAL_DEVICE_CAP_FLAGS is being used in 32 and 16 bits, which one should be used ?

hamishwillee, 2023-04-02:
Probably common.xml.

Given that 14 of the 16 flags are already used you'd have to say more are needed. But given that gimbal is appearing in deployed systems the field that uses these flags in common may no longer be modifiable. Because of this common.xml will be forced to to extend this by using the remaining gimbal bit and having extended flags "at some point".
So common is unlikely to extend the flags, but instead create new ones, and Storm will be left with a bunch of bits that we will never fill.

That is not the ideal of course.

@julianoes Just how "WIP" is gimbal. I was going to talk to you about removing those flags next week. IF it can be modified do we need a bigger field?

FYI @olliw42

julianoes, 2023-04-02:
Interesting. Well, I don't think it's an option to change these messages as they are used by PX4 and ArduPilot, so it would be breaking a lot.

hamishwillee, 2023-04-02:
@patrickelectric So both messages that use this flag are valid in and of themselves - i.e. the field lengths are what they are. The only difference is that common.xml will not be extending that particular enum with new flags, so storm is wasting half the field.
If these were "really" WIP I'd be arguing for the common message to match storm - too late now.

patrickelectric, 2023-04-03:
I was working on https://github.com/mavlink/rust-mavlink and found out this,it's an edge case of the entire dialect and that's sad. It would be nice that we could detect such things before..

hamishwillee, 2023-04-03:
Yes. Would be fairly painful to detect - we'd have to have some kind of validation to check that all fields that use the same enum are the same size and big enough for that enum. Doable, but at some point you have to wonder if worth the coding effort.

olliw42, 2023-04-03:
thx for raising this point.

It's a result/consequence of the recent harmonization of the gimbal **device** messages. Before that the storm32 dialect was using it's own version of gimbal device messages, which had a uint32 for the capabilities flag. Thanks to the harmonization the storm32 gimbal device messages could be removed, since the gimbal device messages didn't incur any loss of capabilities anymore.

With respect to this field, alternatives could have been to, e.g., (a) modify the storm32 gimbal manager messages, which wasn't the preferred option for the same reason why the standard uint16 isn't changed now, or (b) introduce an enum which is factually just exactly the duplicate but has a different name.

> we'd have to have some kind of validation to check that all fields that use the same enum are the same size

this would effectively prevent harmonizations like the one happened here. 

> and big enough for that enum

better :)

I think it is a good thing that this harmonization could happen.

In my defence I want to say that I did bring up all the points, which were missing and which were added now in the course of this harmonization, already ca 2 years ago but were not considered, which led to the storm32 gimbal device version in the first place. In some way it all was avoidable.

I didn't expected that this larger field would cause any troubles or harm. If this would be what you guys would want to happen, I would be perfectly willing to do e.g. (b) (or whatever else non-breaking you could suggest).

patrickelectric, 2023-04-03:
> introduce an enum which is factually just exactly the duplicate but has a different name.

That sounds the best approach for me.

olliw42, 2023-04-03:
many thx for this effort, much appreciated, but it's unfortunately not what it should be

GIMBAL_MANAGER_CAP_FLAGS is not a 32bits version of GIMBAL_DEVICE_CAP_FLAGS, it has many similar fields, but it's not an exact copy.
Second, the STorM32 messages systematically separate between device and manager flags, and here we are concerned with the device flags =>type error (I would think this should make sense especially to the strongly-typisizing folks among us :)). 

one should introduce an enum MAV_STORM32_GIMBAL_DEVICE_CAP_FLAGS, with fields which mirror those of GIMBAL_DEVICE_CAP_FLAGS (but should have names MAV_STORM32_GIMBAL_DEVICE_CAP_FLAGS_XXXX).

if also hamishwillee and whoever else from the mavlink maintainers agrees to this, I will do, but I would want to have that agreement from these maintainers.

from a general perspective I however have to say that settling for the rule "_all fields that use the same enum are ... big enough for that enum_" would make most sense to me ... in which case nothing would have to be done :)

hamishwillee, 2023-04-03:
The larger field does not cause harm in terms of "the standard". From the point of view of MAVLink there is no issue here at all. 
If a particular generator has some kind of technical problem with this then it needs to take a long hard look at itself :-).

IF it were me and I was you @olliw42 I'd do your suggestion of mirroring the common flags in MAV_STORM32_GIMBAL_DEVICE_CAP_FLAGS, and if we add new flags in common, appending them to that enum. Mostly for efficiency of the message. Your call there, unless you want the opinion of @julianoes

hamishwillee, 2023-04-03:
PS THe standard allows for an enum in a dialect to extend an enum or in theory to replace an enum value (though this is disallowed in mavgen). So you might just extend the normal enum in the dialect - it is valid and generators have to cope with this.

julianoes, 2023-04-10:
> one should introduce an enum MAV_STORM32_GIMBAL_DEVICE_CAP_FLAGS, with fields which mirror those of GIMBAL_DEVICE_CAP_FLAGS (but should have names MAV_STORM32_GIMBAL_DEVICE_CAP_FLAGS_XXXX).

That's probably cleanest indeed.

hamishwillee, 2023-04-12:
OK - OK< to leave this with you to update then @olliw42 ?

olliw42, 2023-04-12:
I think the key statement in this thread is 

> The larger field does not cause harm in terms of "the standard". From the point of view of MAVLink there is no issue here at all. If a particular generator has some kind of technical problem with this then it needs to take a long hard look at itself.

and I think this statement should be memorized as what the standard's position is.

Given that, and that you seem to be happy with leaving it up to me, I must admit I'd like to choose what's the easiest for me, namely to just do nothing :D. 

I could argue that it presents a test bed for generator developpers, to learn coping with that a generator needs to take a look at itself, but truth of course is to just not do anything LOL.

I will however, when there is a need for an update of the storm32.xml dialect for other reasons, not forget to consider doing the mirrored enum.

All good?

hamishwillee, 2023-04-13:
Thanks @olliw42 - doing nothing should be fine. There is no "problem" that needs fixing from a MAVLink perspective until/unless we have to extend the flags - at that point that's a "standards" issue, and dialects will need to decide what they do themselves.

@patrickelectric I'm going to close this as "not a defect". If it is an unsurmountable problem for you we will need to discuss further. The reality is  that a dialect can define whatever messages it likes so a generator does need to be able to handle this kind of thing.

closed 2023-04-13 by hamishwillee