mavlink/mavlink #1981: Implicit MAV_FRAME for MAV_CMDs sent as COMMAND_LONG

sypaq-nexton, 2023-04-28:
There are a number of `MAV_CMD_*` messages (such as [`MAV_CMD_DO_SET_ROI_LOCATION`](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_ROI_LOCATION)) that include a `(lat, lon, alt)` location. 
* When sent as a [`COMMAND_INT`](https://mavlink.io/en/messages/common.html#COMMAND_INT), the [`MAV_FRAME`](https://mavlink.io/en/messages/common.html#MAV_FRAME) is included, and the location can be correctly decoded.
* When sent as a [`COMMAND_LONG`](https://mavlink.io/en/messages/common.html#COMMAND_LONG), the `MAV_FRAME` is implicit, and can lead to errors when the sending and receiving systems use a different reference frame (e.g. see https://github.com/ArduPilot/ardupilot/pull/23266).

There are also a number of `MAV_CMD_*` messages (such as [`MAV_CMD_DO_GO_AROUND`](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_GO_AROUND)) that include an altitude, but not altitude reference (either explicitly as is done in [`MAV_CMD_DO_CHANGE_ALTITUDE`](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_CHANGE_ALTITUDE), or implicitly in the description of the altitude field). 

The definitions of the `MAV_CMD_*` messages that include a location or altitude should be updated to explicitly define the `MAV_FRAME` to use when using these messages in a `COMMAND_LONG`.

sypaq-nexton, 2023-04-28:
The other option is to update the [Command Protocol](https://mavlink.io/en/services/command.html) to explicitly state that `MAV_CMD_*` messages that include a location must only be accepted as a `COMMAND_INT`, and must be NACKed (possibly with a new error code in [MAV_RESULT](https://mavlink.io/en/messages/common.html#MAV_RESULT)) when received as a `COMMAND_LONG`.

hamishwillee, 2023-04-28:
@sypaq-nexton Yes, this is a known issue.

The simple fact is that no one should be using `COMMAND_LONG` for positional commands. Firstly because of the loss of precision, but also because, as you noticed, there is no way to set the frame. Worse, there is no way to know the frame that a particular flight stack might be using.

This is bleedingly obvious to me, and yet last time around I could not get traction to have anything stronger than the wording I added here (I might revisit it to also note that the frame cannot be set). I think this is mostly because the flight stacks don't want to have to change code that only handles command_long.

[COMMAND_LONG](https://mavlink.io/en/messages/common.html#COMMAND_LONG)
[image]

[COMMAND_INT](https://mavlink.io/en/messages/common.html#COMMAND_INT)
[image]

I also discussed another option here: https://github.com/mavlink/mavlink/pull/1956#issuecomment-1459068303 - a new message that allows any data in. Sort of stalled on that.

> The other option is to update the [Command Protocol](https://mavlink.io/en/services/command.html) to explicitly state that MAV_CMD_* messages that include a location must only be accepted as a [...]

Again, we're a democracy. You're 100% right, but I need buy in from the flight stacks. This problem is on my TODO list still, and I try revisit regularly. 

We'll leave this open, but don't hold your breath.

sypaq-nexton, 2023-05-01:
@hamishwillee: Thanks for the quick response. 

The issue is that the qualifiers on `COMMAND_INT` and `COMMAND_LONG` only discuss latitude and longitude, not altitude. I've raised PR #1983 to indicate that `MAV_CMD` messages that contain an altitude should be sent as `COMMAND_INT`.

> This is bleedingly obvious to me, and yet last time around I could not get traction [...]

Maybe the interim solution is to provide a mechanism (as discussed by you and @peterbarker [here](https://github.com/ArduPilot/MAVProxy/pull/1149#issuecomment-1441086033) and [here](https://github.com/ArduPilot/MAVProxy/pull/1149#issuecomment-1441109176)) so that receiving systems can progressively indicate their preference for `COMMAND_INT` or `COMMAND_LONG` as they choose, rather than forcing it by changing the current message definitions. I've raised PR #1982 to provide new `MAV_RESULT_COMMAND_LONG_ONLY` and `MAV_RESULT_COMMAND_INT_ONLY` values in the `MAV_RESULT` enum.

hamishwillee, 2023-05-03:
I've responded to your post. I think this idea has merit - certainly just supporting commands in both types does not solve this problem.

Let's see how the discussion in #1982 progresses.

peterbarker, 2026-09-02:
Update on this one.

ArduPilot accepts all commands as both `COMMAND_LONG` and `COMMAND_INT`.

.... *except* when a user chooses to compile `COMMAND_LONG` support out entirely!  In the case that `COMMAND_LONG` is compiled out we return a COMMAND_ACK with `MAV_RESULT_COMMAND_INT_ONLY` in it.  This does mean that any command which does transport floats in p5/p6 won't work well - but that's just not a problem for AP at the moment

ArduPilot converts all command-long to command-int before executing them, and thus has an explicit list of location-containing messages which it declares a frame for.  These can be found by grepping for `GCS_MAVLINK.*::mav_frame_for_command_long` (it differs vehicle-to-vehicle, sadly)

We have made no plans to *stop* supporting these location commands via `COMMAND_LONG`, which is what #1982 implies.

nexton-winjeel, 2026-09-03:
@peterbarker:
> ArduPilot converts all command-long to command-int before executing them, and thus has an explicit list of location-containing messages which it declares a frame for. These can be found by grepping for GCS_MAVLINK.*::mav_frame_for_command_long (it differs vehicle-to-vehicle, sadly)

Ack. The point of this issue is to try to make these `MAV_FRAME`s explicit in the interface definition, rather than being implementation defined.

hamishwillee, 2026-09-03:
> We have made no plans to _stop_ supporting these location commands via `COMMAND_LONG`, which is what [#1982](https://github.com/mavlink/mavlink/pull/1982) implies.

@peterbarker As you know, a the goal of MAVLink is to provide predictable component-agnostic behaviour, so I'd say this is implicitly non-compliant with the intent and spirit of the standard. 

To be a good MAVLink citizen you either have to NACK ... or we need to define default frames for when used in `LONG`. If you can confirm that the main stacks use the same default value we can add that information.

FYI PX4 is currently looking at NACK for at least some of these cases.

peterbarker, 2026-09-04:
> [@peterbarker](https://github.com/peterbarker) As you know, a the goal of MAVLink is to provide predictable component-agnostic behaviour, so I'd say this is implicitly non-compliant with the intent [...]

Sure, if you consider the standard post-addition of those two new command responses!  Prior to that the stack had to make assumptions on what the caller meant.  The patches that gave us COMMAND_INT support for the relevant commands were crafted to allow for as much code reuse as possible (thus teh convert-to-command-long aspect of it).
 
> To be a good MAVLink citizen you either have to NACK ... 

Problem there is that a lot of the commands are very long-standing.  Moving people forward is often best accomplished by providing a better alternative.  So https://github.com/ArduPilot/ardupilot/pull/34259 allows ArduCopter to take off relative to current altitude rather than home.  Since our `COMMAND_LONG` handling is fixed at "above home", this provides a solid reason for ground control stations to move to using the command-int version!

> or we need to define default frames for when used in `LONG`. If you can confirm that the main stacks use the same default value we can add that information.

Happy to do so.  But `takeoff` being in a different frame PX4-vs-ArduPilot is not a good start :-)

`MAV_CMD_DO_SET_GLOBAL_ORIGIN` would be a safe one to start the list with - that *must* be AMSL.

> FYI PX4 is currently looking at NACK for at least some of these cases.

Friendlier would be to accept as either, as AP did.  Note that QGC currently uses command-long for takeoff on PX4.... and uses AMSL for its alt field vs relative in ArduPilot.

I'm going to look at weening ArduPilot off the command-long versions somehow.  

Currently when converting from COMMAND_LONG to COMMAND_INT ArduPilot assumes the following frames:

| Command | Assumed frame | Where |
|---|---|---|
| `MAV_CMD_FIXED_MAG_CAL_YAW` | `MAV_FRAME_GLOBAL_RELATIVE_ALT` | GCS_Common |
| `MAV_CMD_DO_SET_ROI` | `MAV_FRAME_GLOBAL_RELATIVE_ALT` | GCS_Common |
| `MAV_CMD_DO_SET_ROI_LOCATION` | `MAV_FRAME_GLOBAL_RELATIVE_ALT` | GCS_Common |
| `MAV_CMD_DO_SET_HOME` | `MAV_FRAME_GLOBAL` | GCS_Common |
| `MAV_CMD_NAV_TAKEOFF` | `MAV_FRAME_GLOBAL_RELATIVE_ALT` | Copter, Blimp |
| `MAV_CMD_NAV_VTOL_TAKEOFF` | `MAV_FRAME_GLOBAL_RELATIVE_ALT` | Copter |

(Plane converts a COMMAND_LONG `NAV_TAKEOFF` to an explicit `MAV_FRAME_LOCAL_OFFSET_NED` and only accepts that frame, so it is not a guess there, but the warning still fires.)

An audit of how the major GCSs currently send these commands (as GCS commands, not as mission items):

| Command | MAVProxy | QGroundControl | MissionPlanner |
|---|---|---|---|
| `FIXED_MAG_CAL_YAW` | COMMAND_LONG | COMMAND_LONG | COMMAND_LONG |
| `DO_SET_ROI` | not sent | not sent | COMMAND_INT |
| `DO_SET_ROI_LOCATION` | COMMAND_INT | COMMAND_INT (COMMAND_LONG fallback if the vehicle lacks the COMMAND_INT capability bit) | COMMAND_INT |
| `DO_SET_HOME` | mixture (map "Set Home" and movinghome use COMMAND_INT; map "Set Home (with height)" uses COMMAND_LONG) | COMMAND_LONG | COMMAND_INT |
| `NAV_TAKEOFF` | mostly COMMAND_LONG (`takeoff` command, swarm); chat module uses COMMAND_INT | COMMAND_LONG | COMMAND_LONG |
| `NAV_VTOL_TAKEOFF` | not sent | not sent | not sent |

"not sent" means the GCS has no code path that sends the command as a COMMAND_LONG or COMMAND_INT; it may still emit it as a mission item.

### For comparison: how PX4 handles the same commands

PX4 has accepted COMMAND_INT generically since August 2014 (PX4 commit 0d87dc9923, first shipped in v1.0.0), and has advertised `MAV_PROTOCOL_CAPABILITY_COMMAND_INT` since 2015.  COMMAND_LONG and COMMAND_INT are both flattened into the same internal `vehicle_command` uORB message in `mavlink_receiver.cpp`; that message has no frame field, so every command handler sees identical data regardless of which MAVLink message carried it.  Any command PX4 handles at all is therefore accepted via both COMMAND_LONG and COMMAND_INT from the day its handler was added (or from August 2014 for the older handlers).

PX4 never guesses a frame because it never honours one.  From 2014 until May 2026 the COMMAND_INT `frame` field was ignored entirely and x/y were unconditionally divided by 1e7.  Since May 2026 (PX4 commit bf1e591aab, PR #27335, first in v1.18.0-beta1) local frames divide x/y by 1e4 instead, but the handlers still interpret the result as latitude/longitude, so in practice only global frames are meaningful.  Altitude is taken as absolute (AMSL) everywhere; a `MAV_FRAME_GLOBAL_RELATIVE_ALT` COMMAND_INT is treated exactly like `MAV_FRAME_GLOBAL`.  In ArduPilot terms, PX4 hard-codes `MAV_FRAME_GLOBAL` for every location-carrying command on both message types.

PX4 also rejects a COMMAND_LONG whose param5/param6 are both INT32_MAX, and a COMMAND_INT whose x/y are both the NaN bit pattern, with `MAV_RESULT_DENIED` as probable encoding mistakes.

| Command | COMMAND_LONG | COMMAND_INT | Frame assumed (both message types) | Handler since | COMMAND_INT accepted since |
|---|---|---|---|---|---|
| `FIXED_MAG_CAL_YAW` | yes | yes | n/a: lat/lon read from param3/param4 per the spec; x/y/z and frame unused | Sept 2020 (v1.12.0) | Sept 2020 |
| `DO_SET_ROI` | yes | yes | n/a: only param1 (mode) is used; lat/lon/alt ignored | Sept 2017 (Navigator) | Sept 2017 |
[... cut ...]
| `NAV_TAKEOFF` | yes | yes | `GLOBAL`: alt (param7/z) AMSL; lat/lon optional, NaN or INT32_MAX means current position | Apr 2013 | Aug 2014 (v1.0.0) |
| `NAV_VTOL_TAKEOFF` | yes | yes | `GLOBAL`: alt (param7/z) is the absolute transition altitude; lat/lon is the post-transition loiter position | Feb 2022 (v1.13.0) | Feb 2022 |

Note that ArduPilot's own `FIXED_MAG_CAL_YAW` handler also reads lat/lon from param3/param4, so the `frame_map` entry for it is never actually consulted (`command_long_stores_location()` does not list it).

peterbarker, 2026-09-16:
I've created https://github.com/ArduPilot/ardupilot/pull/34420 as a start to weening ArduPilot off the "implicit frame" issue.

@hamishwillee PX4-AutoPilot really, really should start rejecting commands with frames it doesn't cope with for that command.  https://github.com/PX4/PX4-Autopilot/pull/26408 was a touch blunt, but I will note that the PR put in to start fixing things is languishing since #26408 was reverted!

hamishwillee, 2026-09-17:
@peterbarker Thanks for all that detailed analysis above. I haven't forgotten it, I'm just trying to stay afloat. 

> [@hamishwillee](https://github.com/hamishwillee) PX4-AutoPilot really, really should start rejecting commands with frames it doesn't cope with for that command. [PX4/PX4-Autopilot#26408](https://github.com/PX4/PX4-Autopilot/pull/26408) was a touch blunt, but I will note that the PR put in to start fixing things is languishing since #26408 was reverted!

Absolutely. The TLDR is that we are doing this but it is waiting on @julianoes to do a first example in https://github.com/PX4/PX4-Autopilot/pull/27280 which we will then use as the basis of a pattern to roll out everywhere else.

This falls out of earlier work we did to reject commands and mission items that pass non-sentinel values in unsupported params and this PR https://github.com/PX4/PX4-Autopilot/pull/28372 which attempted to fix  a frame issue. As per https://github.com/PX4/PX4-Autopilot/pull/28372#issuecomment-5519278963 we realize we really have to do this check at point of exectution and have a phased introduction.

With respect to the implicit frame issue, also agree. Let's try keep in sync - here's the PX4 version https://github.com/PX4/PX4-Autopilot/pull/28733