mavlink/mavlink #1985: Not able to extract information from wireshark captured Mavlink protocol packet

kryptoniancode, 2023-05-11:
I have used this to generate Wireshark lua plugin for Mavlink, then captured the packets.
```
python3 -m pymavlink.tools.mavgen --lang=WLua --wire-protocol=2.0 --output=mavlink_2_common message_definitions/v1.0/common.xml
```

[image]

But using this python code does not give any information about `mavlink_proto`. What could be changed to get the information?
```
import pyshark

capture = pyshark.FileCapture('mavlink/mission.pcapng', display_filter='mavlink_proto')
capture.set_debug()

for packet in capture:
    src_ip = packet.ip.src
    dst_ip = packet.ip.dst
    src_port = packet.udp.srcport
    dst_port = packet.udp.dstport
    mavlink_payload = packet.mavlink.payload

    print(f'Source IP: {src_ip}, Destination IP: {dst_ip}, '
          f'Source Port: {src_port}, Destination Port: {dst_port}, '
          f'Mavlink Payload: {mavlink_payload}')

capture.close()
```

Error
```
pyshark.capture.capture.TSharkCrashException: TShark (pid 12497) seems to have crashed (retcode: 2).
Last error line: tshark: "mavlink_proto" is neither a field nor a protocol name.
```

What could be better way to extract the info?

kryptoniancode, 2023-05-18:
I have used this command to generated lua plugin and edited last line to have TCP and port 5760.

```
python3 -m pymavlink.tools.mavgen --lang=WLua --wire-protocol=2.0 --output=mavlink_2_common message_definitions/v1.0/common.xml
```

hamishwillee, 2023-05-18:
Yes, but common.xml doesn't have the message ATTITUDE - right? So where is that coming from? And how do you expect the plugin to deconstruct a message for which it does not have a definition?

kryptoniancode, 2023-05-18:
Okay, I understand.

Any way I have used scapy to dissect the raw packet and it working now. I have not used mavlink lua to capture instead captured raw packet and used scapy to get information.

closed 2023-05-18 by hamishwillee