mavlink/mavlink #1987: mavlink-gbp-release fails to build after version 2023.5.5-1

bromeara, 2023-05-16:
Hello,
## Error
While building mavlink from source on the branch release/melodic/mavlink/2023.5.5-1 I got the following error.
```
CMake Error: File /home/dev/Catkin/src/mavlink/config.h.in does not exist.
CMake Error at CMakeLists.txt:68 (configure_file):
  configure_file Problem configuring file
```
This was caused by the missing file `config.h.in`, which seems to have been removed in commit [b0d9d52](https://github.com/mavlink/mavlink-gbp-release/commit/b0d9d52513c42beac15f6784ffee13fdd7e28ac0). While I did not try to build previous branches, I did verify that this file was also removed in release/noetic/mavlink/2023.5.5-1 .

## Solutions
I am not sure if this file was removed on purpose, as it is just the line `#define MAVLINK_VERSION "${PROJECT_VERSION}"`
This issue can be fixed by:
1. adding the file back into the directory
or 
2. removing lines [ 67-69](https://github.com/mavlink/mavlink-gbp-release/blob/fba004046bc4b64746dec439a97ab929664bd2fc/CMakeLists.txt#L67-L69) from the CMakeLists.txt

Since I think everything for mavlink-gbp-release is auto-generated the solution may be a bit more complicated, but I went the route of adding the file back into my directory since if nothing uses it anymore there shouldn't be any harm, but if this will cause undesired side effects please let me know.

mortenfyhn, 2023-06-12:
I have the same problem, would be wonderful to have this resolved for the next release.

mortenfyhn, 2023-06-12:
Aha! It's just not on the Melodic release, which I assume is because ROS Melodic is past end of life now?

vooon, 2023-06-12:
Yep. But recently EOLed Melodic and Foxy should be reverted to Jan release.

mortenfyhn, 2023-06-13:
Yeah, that makes sense. Thank you for clarifying!

hamishwillee, 2023-06-14:
Closing - not a MAVLInk project issue.

closed 2023-06-14 by hamishwillee