mavlink/mavlink #1997: Move enums in description to real enums

patrickelectric, 2023-05-25:
E.g: `MAV_CMD_NAV_WAYPOINT`, has on the first field:

`<param index="1" label="Speed Type" minValue="0" maxValue="3" increment="1">Speed type (0=Airspeed, 1=Ground Speed, 2=Climb Speed, 3=Descent Speed)</param>`

The speed type should be defined in an enum, and not being used in the description.

Doing so will empower the tools that use the specification to also provide possible valid values without parsing the description (that's just a text without xml validation).

hamishwillee, 2023-05-25:
We often don't want to clutter the namespace if there are only a few values to use and they are unlikely to be extended. On the other hand, enums make it clear than you mean "this value" and prevents flight stacks from doing things like assuming values will not change and condition of greater-than.

I'd be happy with this. Do you want to PR it?

hamishwillee, 2023-05-31:
@patrickelectric ^^^^ ?

patrickelectric, 2023-06-05:
Hi @hamishwillee, sorry, I'm in vacation, should be back on this before the end of the month.

hamishwillee, 2023-09-14:
Cool, should be fixed by https://github.com/mavlink/mavlink/pull/2039

closed 2023-09-27 by hamishwillee

patrickelectric, 2023-09-27:
Hi @hamishwillee, sorry for not answering it before!

As said, make them as enum help us to have clear behavior and intention. Hiding enums under ranges makes it harder to define and make it clear for the user and tools how the values are used.

hamishwillee, 2023-09-27:
No worries. All good. Merged!

PR #2039 (2023-09-27): SPEED_TYPE enum for MAV_CMD_DO_CHANGE_SPEED
This adds enum for speed types that can be set in [MAV_CMD_DO_CHANGE_SPEED](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_CHANGE_SPEED).

Fixes #1997