mavlink/mavlink #2018: Generateing Swift library fails with compile error

valeriyvan, 2023-07-12:
```
python3 -m pymavlink.tools.mavgen --lang=Swift --wire-protocol=2.0 --output=generated/swift/mavlink/v2.0 message_definitions/v1.0/all.xml
Validating message_definitions/v1.0/all.xml
Parsing message_definitions/v1.0/all.xml
Validating /Users/*****/DronePro/mavlink/message_definitions/v1.0/ardupilotmega.xml
Parsing /Users/*****/DronePro/mavlink/message_definitions/v1.0/ardupilotmega.xml
Validating /Users/*****/DronePro/mavlink/message_definitions/v1.0/ASLUAV.xml
Parsing /Users/*****/DronePro/mavlink/message_definitions/v1.0/ASLUAV.xml
Validating /Users/*****/DronePro/mavlink/message_definitions/v1.0/common.xml
Parsing /Users/*****/DronePro/mavlink/message_definitions/v1.0/common.xml
Validating /Users/*****/DronePro/mavlink/message_definitions/v1.0/development.xml
Parsing /Users/*****/DronePro/mavlink/message_definitions/v1.0/development.xml
Validating /Users/*****/DronePro/mavlink/message_definitions/v1.0/icarous.xml
Parsing /Users/*****/DronePro/mavlink/message_definitions/v1.0/icarous.xml
Validating /Users/*****/DronePro/mavlink/message_definitions/v1.0/minimal.xml
Parsing /Users/*****/DronePro/mavlink/message_definitions/v1.0/minimal.xml
Validating /Users/*****/DronePro/mavlink/message_definitions/v1.0/python_array_test.xml
Parsing /Users/*****/DronePro/mavlink/message_definitions/v1.0/python_array_test.xml
Validating /Users/*****/DronePro/mavlink/message_definitions/v1.0/standard.xml
Parsing /Users/*****/DronePro/mavlink/message_definitions/v1.0/standard.xml
Validating /Users/*****/DronePro/mavlink/message_definitions/v1.0/test.xml
Parsing /Users/*****/DronePro/mavlink/message_definitions/v1.0/test.xml
Validating /Users/*****/DronePro/mavlink/message_definitions/v1.0/ualberta.xml
Parsing /Users/*****/DronePro/mavlink/message_definitions/v1.0/ualberta.xml
Validating /Users/*****/DronePro/mavlink/message_definitions/v1.0/uAvionix.xml
Parsing /Users/*****/DronePro/mavlink/message_definitions/v1.0/uAvionix.xml
Validating /Users/*****/DronePro/mavlink/message_definitions/v1.0/storm32.xml
Parsing /Users/*****/DronePro/mavlink/message_definitions/v1.0/storm32.xml
Validating /Users/*****/DronePro/mavlink/message_definitions/v1.0/AVSSUAS.xml
Parsing /Users/*****/DronePro/mavlink/message_definitions/v1.0/AVSSUAS.xml
Validating /Users/*****/DronePro/mavlink/message_definitions/v1.0/cubepilot.xml
Parsing /Users/*****/DronePro/mavlink/message_definitions/v1.0/cubepilot.xml
Validating /Users/*****/DronePro/mavlink/message_definitions/v1.0/csAirLink.xml
Parsing /Users/*****/DronePro/mavlink/message_definitions/v1.0/csAirLink.xml
Merged enum MAV_CMD
Merged enum MAV_CMD
Merged enum MAV_CMD
Merged enum MAV_CMD
Merged enum MAV_CMD
Merged enum MAV_CMD
Found 355 MAVLink message types in 16 XML files
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/Users/*****/DronePro/mavlink/pymavlink/tools/mavgen.py", line 31, in <module>
    mavgen.mavgen(args, args.definitions)
  File "/Users/*****/DronePro/mavlink/pymavlink/generator/mavgen.py", line 291, in mavgen
    mavgen_swift.generate(opts.output, xml)
  File "/Users/*****/DronePro/mavlink/pymavlink/generator/mavgen_swift.py", line 376, in generate
    generate_messages_type_info(msgs, enums)
  File "/Users/*****/DronePro/mavlink/pymavlink/generator/mavgen_swift.py", line 347, in generate_messages_type_info
    field.initial_value = "try data." + swift_types[field.type][2] % field.wire_offset
                                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
TypeError: not enough arguments for format string
```

hamishwillee, 2023-07-13:
Can you please cross post this to https://github.com/ArduPilot/pymavlink/issues

I could do this, but then you won't get notifications, and that is where any work will happen.

valeriyvan, 2023-07-13:
> Can you please cross post this to https://github.com/ArduPilot/pymavlink/issues
> 
> I could do this, but then you won't get notifications, and that is where any work will happen.

https://github.com/ArduPilot/pymavlink/issues/841

hamishwillee, 2026-08-25:
Closing as this is being addressed in https://github.com/ArduPilot/pymavlink/issues/841

If this is an important use case suggest engaging on that issue to find solutions.

closed 2026-08-25 by hamishwillee