mavlink/mavlink #2022: MANUAL_CONTROL has "target" field rather than "target_system"

JMare, 2023-07-24:
I noticed this after deep diving a bug with using joysticks on multiple vehicles from QGroundcontrol. 
https://github.com/mavlink/qgroundcontrol/issues/10544

The MANUAL_CONTROL message has a field called "target" and I believe it is supposed to be called "target_system". All other messages seem to use "target_system".
I found this because MAVProxy was not correctly routing these packets, and after digging into the code I found it was because it was looking for 'target_system' which is not present in this message.

Relevant code in MAVProxy where the 'target_system' attribute is pulled out.
https://github.com/ArduPilot/MAVProxy/blob/85806b9561814402868434b56848b060cd12c7f5/MAVProxy/mavproxy.py#L860

hamishwillee, 2023-07-27:
Thanks. You are correct, but there is not much we can do about it.

The use of target_system and target_component to identify the target of a message is a convention; generated libraries provide helper methods that allow you to more easily extract the values of fields with these names. If not defined, again by convention, the target component/system are assumed 0, which means "broadcast". There has been much discussion over the years about whether these really should have been part of the packet header (I think so).

Even though this does not match the convention there is not much that can be done in MAVLink - if we change the name of that field it will break every single existing implementation that uses this message, most of which will have already adapted to it.

The only short term alternative is that we add target_system as an extension (losing any truncation) and populate both fields). I'm not sure this works - if the value is zero a system that has been updated would still need to check if the original value was non-zero (because if the command was sent by a non-updated system it would always be zero).

We have talked of a new MANUAL_CONTROL message that omits the buttons, since mostly these can be achieved better by directly sending MAVLink commands for actions. However that will also be problematic to get through.

I'm going to close this as "Not going to fix", but you're welcome to come back with a PR if you can get agreement that either ArduPilot or PX4  will support a change.

closed 2023-07-27 by hamishwillee

JMare, 2023-07-27:
@hamishwillee thanks for the detailed explanation.
Understood why this can't be easily fixed without silently breaking things in introspective languages.

I will look to raise on the mavproxy side, since part of the issue may be that mavproxy is not following the convention that messages without target_system should be treated as broadcast. (it appears that mavproxy is treading messages without target_system as directed only to the active vehicle)
The bug I was chasing is an edge case anyway which will probably only appear in simulation environments.

Thanks

hamishwillee, 2023-07-27:
> Understood why this can't be easily fixed without silently breaking things in introspective languages.

It's actually something that will affect any language. MAVLink packets have a CRC "EXTRA" applied over the message _definition_ and sent in the packet overlaid over the normal CRC. This is what tells the recipient of a message that their definition is the same as the senders. The name of a field is part of this message definition, so we can't update the message without breaking all existing users.

W.r.t. routing, almost no setups fully follow routing rules. It's a pity but the authors of the various flight stacks and GCS are pragmatic, so there might be a good reason in the particular case.

JMare, 2023-07-27:
Ahh yes, I hadn't considered or fully understood CRC EXTRA before, that explains a lot.