mavlink/mavlink #2028: Encountered udp connection issue while using the mavlink library

JiaXu-Zhang, 2023-08-15:
Encountered udp connection issue while using the mavlink library
I can sent heartbeat but I cannot sent pwm value to rc_channels, what should I do ? thx ！
`    mavlink_msg_rc_channels_override_pack_chan(message.sysid, message.compid,MAVLINK_COMM_0, &message,1,0,1900, 1500, 1500, 1500, 1500, 1500, 1000, 1000,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500);
    `
and read 
`case MAVLINK_MSG_ID_RC_CHANNELS:
                mavlink_rc_channels_t rc_channels;
                mavlink_msg_rc_channels_decode(&message, &rc_channels);
                std::cout << "RC Channel 1 Raw Value: " << rc_channels.chan1_raw<< std::endl;`
[udp_example.txt](https://github.com/mavlink/mavlink/files/12341439/udp_example.txt)
and complete code

hamishwillee, 2023-08-16:
You say that you can send a heartbeat but not a message to set RC channels. How do you know that you are successfully sending a heartbeat and not successfully sending the command to set RC channels? Is the code crashing when you try send the second message?

Assuming you're saying you can **receive** a heartbeat then I assume you have a connection. Have you checked that your flight stack supports the command you are trying to send? You haven't said what flight stack you are using so it is impossible for me to comment.

If I had to do this I would start by using a Python script and verify that the flight stack does the right thing with known-good code. YOu might start from https://discuss.bluerobotics.com/t/pymavlink-and-rc-override-for-autonomous-guidance-in-acro-mode/2401

Hope that helps. The next step would be to raise this request on the discussion forums of the flight stack that you are using. They will be able to help you with flight-stack specific issues. You should confirm what version of the flight stack you are testing against.

closed 2023-08-16 by hamishwillee

JiaXu-Zhang, 2023-08-16:
> 
Thanks for your reply,I use this in SITL, "arm_sendrc.txt"use pymavlink to arm throttle and send rc pwm value , it can work,
[arm_sendrc.txt](https://github.com/mavlink/mavlink/files/12356195/arm_sendrc.txt)
[image]
I also can arm/disarm in C++ ,but I still can not send rc value 
[image]
this is complete code

[arm_disarm.txt](https://github.com/mavlink/mavlink/files/12356242/arm_disarm.txt)

JiaXu-Zhang, 2023-08-16:
@hamishwillee

hamishwillee, 2023-08-16:
@JiaXu-Zhang That answers my question - you're working on ArduPilot - which does indeed support this feature. Great to see that Pymavlink works too.

So that means there is something wrong with your C code. Unfortunately I am not great with C - please raise this question on the ArduPilot forums. I am sure you will find code that does this.

JiaXu-Zhang, 2023-08-16:
Thank you for your suggestion @hamishwillee