mavlink/mavlink #2033: CUSTOM mavlink messages - BAD_DATA / CRC error

Mohnish4533, 2023-08-18:
From the senders side the message sent is : ['fe', '2', '1', 'ff', '0', '1', '1', '2', '53', 'd5']
But on the receivers side the message received is : BAD_DATA {invalid MAVLink CRC in msgID 1 0xd553 should be 0xd920, data:['fe', '2', '1', 'ff', '0', '1', '1', '2', '53', 'd5']

What I've tried:
> tried generating the CUSTOM script in both MAVLink1 and MAVLink2, but the error still persisted.

After a bit more of digging I came across a possible cause of this error. The cuase i came across is related to CRC_EXTRA not working for custom mavlink message implementations. 

Has anyone encountered the same issue? It would really be helpful if someone could help out.

Thanks in advance!

hamishwillee, 2023-08-23:
The message means you have a different CRC_EXTRA calculated on sender and receiver end for the same message. That implies your sender and receiver have been built with a different definition of the messages. Note that the CRC depends on the message name, and the base field name and type, in sending order.

The fact that you are using a custom message should not matter.

hamishwillee, 2023-09-05:
Closing. If you need to discuss further suggest use discussion boards as I am pretty sure this is a library mismatch.

closed 2023-09-05 by hamishwillee