mavlink/mavlink #2040: CI is failing Node tests

magicrub, 2023-09-26:
CI is failing Node tests. Seems unrelated to anything from this repo, suspect a Node library dependency just expired/retired/exploded

hamishwillee, 2023-09-26:
The issue in node 12 test is the use of `descriptor?.isOwn` - the optional chaining operator appears first in node 14 https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Operators/Optional_chaining

We should drop that. Not sure about the other one.

hamishwillee, 2023-09-27:
@julianoes Any ideas on the Node 14 failure? Copied from [here](https://github.com/mavlink/mavlink/actions/runs/6179806789/job/16775320760?pr=2039) (starting from first error):
<details>
Merged enum MAV_CMD
Merged enum MAV_CMD
Merged enum MAV_CMD
Merged enum MAV_CMD
ERROR: Duplicate enum names:
	SPEED_TYPE.SPEED_TYPE_AIRSPEED = 0 @ /home/runner/work/mavlink/mavlink/message_definitions/v1.0/common.xml:3438
	SPEED_TYPE.SPEED_TYPE_AIRSPEED = 0 @ /home/runner/work/mavlink/mavlink/message_definitions/v1.0/ardupilotmega.xml:27
rm: cannot remove './javascript/implementations/mavlink_all_v1.0/local_modules': No such file or directory
Validating ../../message_definitions/v1.0/ardupilotmega.xml
Parsing ../../message_definitions/v1.0/ardupilotmega.xml
Validating /home/runner/work/mavlink/mavlink/message_definitions/v1.0/common.xml
Parsing /home/runner/work/mavlink/mavlink/message_definitions/v1.0/common.xml
Validating /home/runner/work/mavlink/mavlink/message_definitions/v1.0/uAvionix.xml
Parsing /home/runner/work/mavlink/mavlink/message_definitions/v1.0/uAvionix.xml
Validating /home/runner/work/mavlink/mavlink/message_definitions/v1.0/icarous.xml
Parsing /home/runner/work/mavlink/mavlink/message_definitions/v1.0/icarous.xml
Validating /home/runner/work/mavlink/mavlink/message_definitions/v1.0/cubepilot.xml
Parsing /home/runner/work/mavlink/mavlink/message_definitions/v1.0/cubepilot.xml
Validating /home/runner/work/mavlink/mavlink/message_definitions/v1.0/csAirLink.xml
Parsing /home/runner/work/mavlink/mavlink/message_definitions/v1.0/csAirLink.xml
Validating /home/runner/work/mavlink/mavlink/message_definitions/v1.0/standard.xml
Parsing /home/runner/work/mavlink/mavlink/message_definitions/v1.0/standard.xml
Validating /home/runner/work/mavlink/mavlink/message_definitions/v1.0/minimal.xml
Parsing /home/runner/work/mavlink/mavlink/message_definitions/v1.0/minimal.xml
Merged enum MAV_CMD
Merged enum SPEED_TYPE
ERROR: Duplicate enum names:
	SPEED_TYPE.SPEED_TYPE_AIRSPEED = 0 @ /home/runner/work/mavlink/mavlink/message_definitions/v1.0/common.xml:3438
	SPEED_TYPE.SPEED_TYPE_AIRSPEED = 0 @ ../../message_definitions/v1.0/ardupilotmega.xml:27
rm: cannot remove './javascript/implementations/mavlink_ardupilotmega_v1.0/local_modules': No such file or directory
Validating ../../message_definitions/v1.0/common.xml
Parsing ../../message_definitions/v1.0/common.xml
Validating /home/runner/work/mavlink/mavlink/message_definitions/v1.0/standard.xml
Parsing /home/runner/work/mavlink/mavlink/message_definitions/v1.0/standard.xml
Validating /home/runner/work/mavlink/mavlink/message_definitions/v1.0/minimal.xml
Parsing /home/runner/work/mavlink/mavlink/message_definitions/v1.0/minimal.xml
Found 142 MAVLink message types in 3 XML files
Generating javascript/implementations/mavlink_common_v1.0/mavlink.js
Generating preamble
Generating enums
Generating message IDs
Generating class definitions
Generating MAVLink class
Generated javascript/implementations/mavlink_common_v1.0/mavlink.js OK
Generating TESTS javascript/implementations/mavlink_common_v1.0/mavlink.tests.js
Generating preamble
Generating MAVLink class
Generating TESTS javascript/implementations/mavlink_common_v1.0/mavlink.tests.js
rm: cannot remove './javascript/implementations/mavlink_common_v1.0/local_modules': No such file or directory
Validating ../../message_definitions/v1.0/csAirLink.xml
Parsing ../../message_definitions/v1.0/csAirLink.xml
Found 0 MAVLink message types in 1 XML files
Generating javascript/implementations/mavlink_csAirLink_v1.0/mavlink.js
Generating preamble
Generating enums
Generating message IDs
Generating class definitions
Generating MAVLink class
Generated javascript/implementations/mavlink_csAirLink_v1.0/mavlink.js OK
Generating TESTS javascript/implementations/mavlink_csAirLink_v1.0/mavlink.tests.js
Generating preamble
Generating MAVLink class
Generating TESTS javascript/implementations/mavlink_csAirLink_v1.0/mavlink.tests.js
rm: cannot remove './javascript/implementations/mavlink_csAirLink_v1.0/local_modules': No such file or directory
Validating ../../message_definitions/v1.0/cubepilot.xml
Parsing ../../message_definitions/v1.0/cubepilot.xml
Validating /home/runner/work/mavlink/mavlink/message_definitions/v1.0/common.xml
Parsing /home/runner/work/mavlink/mavlink/message_definitions/v1.0/common.xml
[... 774 lines cut ...]
       "before each" hook for "Can pack itself":
     Error: Cannot find module '../implementations/mavlink_ardupilotmega_v2.0/mavlink.js'
Require stack:
- /home/runner/work/mavlink/mavlink/pymavlink/generator/javascript/test/message20.js
      at Function.Module._resolveFilename (internal/modules/cjs/loader.js:931:15)
      at Function.Module._load (internal/modules/cjs/loader.js:774:27)
      at Module.require (internal/modules/cjs/loader.js:1003:19)
      at require (internal/modules/cjs/helpers.js:107:18)
      at Context.<anonymous> (test/message20.js:[338](https://github.com/mavlink/mavlink/actions/runs/6179806789/job/16775320760?pr=2039#step:7:339):47)
      at processImmediate (internal/timers.js:464:21)

  6) MAVLink 2.0 message
       "before each" hook for "has a set function to facilitate vivifying the object":
     Error: Cannot find module '../implementations/mavlink_ardupilotmega_v2.0/mavlink.js'
Require stack:
- /home/runner/work/mavlink/mavlink/pymavlink/generator/javascript/test/message20.js
      at Function.Module._resolveFilename (internal/modules/cjs/loader.js:931:15)
[... cut ...] js:774:27)
      at Module.require (internal/modules/cjs/loader.js:1003:19)
      at require (internal/modules/cjs/helpers.js:107:18)
      at Context.<anonymous> (test/message20.js:[351](https://github.com/mavlink/mavlink/actions/runs/6179806789/job/16775320760?pr=2039#step:7:352):47)
      at processImmediate (internal/timers.js:[464](https://github.com/mavlink/mavlink/actions/runs/6179806789/job/16775320760?pr=2039#step:7:465):21)

npm ERR! Test failed.  See above for more details.
</details>

magicrub, 2023-09-27:
PR https://github.com/mavlink/mavlink/pull/2041 is also failing Node 16.

hamishwillee, 2023-09-27:
> PR https://github.com/mavlink/mavlink/pull/2041 is also failing Node 16.

Yeah, but for whatever reason this did fix node 14! 

I've added 15, 17 to that PR too to see if it fails in the same way.

magicrub, 2023-09-27:
[image]

Not looking good...

hamishwillee, 2023-09-27:
HOld on, I wonder if the 14 error is caused by 

```
ERROR: Duplicate enum names:
	SPEED_TYPE.SPEED_TYPE_AIRSPEED = 0 @ /home/runner/work/mavlink/mavlink/message_definitions/v1.0/common.xml:3464
	SPEED_TYPE.SPEED_TYPE_AIRSPEED = 0 @ /home/runner/work/mavlink/mavlink/message_definitions/v1.0/ardupilotmega.xml:27
```

We just added SPEED_TYPE_AIRSPEED as an enum but I see it is already in ArduPilot. I think I need to delete that from the upstream Ardupilot. I am disappointed that removal didn't trigger an error.

hamishwillee, 2023-09-27:
Try to fix that case in https://github.com/mavlink/mavlink/pull/2042

hamishwillee, 2026-08-25:
CI upated and errors no longer valid.

closed 2026-08-25 by hamishwillee