mavlink/mavlink #2043: Every geofences with its own action

Igor-Misic, 2023-10-09:
Can we add this feature to geofences? (@hamishwillee I have seen your comment)

>    Every geofence in a plan can have its own action;

https://github.com/mavlink/mavlink/blob/b295e70eefa5eb4eee4cf6531e33f4b4c4e472d6/message_definitions/v1.0/development.xml#L200

I was thinking of adding action to param7 here:
https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml#L2340
https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml#L2351
https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml#L2362
https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml#L2373

With this, we can select different actions in QGC and then send them to PX4. When geofence is breached we can do different actions.

hamishwillee, 2023-10-11:
Hi @Igor-Misic 

The command [MAV_CMD_SET_FENCE_BREACH_ACTION](https://mavlink.io/en/messages/development.html#MAV_CMD_SET_FENCE_BREACH_ACTION) is supposed to do what you need - i.e. it is **how** you add an action for each separate fence.

> As part of a mission protocol plan it sets the fence action for the _next complete geofence definition_ *after* the command. 
> - Note: A fence action defined in a plan will override the default system setting (even if the system-default is [FENCE_ACTION_NONE](https://mavlink.io/en/messages/development.html#FENCE_ACTION_NONE)
> - Note: Every geofence in a plan can have its own action; if no fence action is defined for a particular fence the system-default will be used. 
> - Note: The flight stack should reject a plan or command that uses a geofence action that it does not support and send a STATUSTEXT with the reason.

Basically you add this item before you declare a circular or polygon fence and it applies only to the next fence. If you have a fence definition made up of a bunch of [MAV_CMD_NAV_FENCE_POLYGON_VERTEX_INCLUSION](https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_FENCE_POLYGON_VERTEX_INCLUSION) you would declare this once and it would apply for the whole fence that follows, but not (say) the next circular fence.

I think this is better than using a reserved part of a vertix definition because a polygon is made up of multiple vertices, and the rules would necessarily be more complex.

No one has implemented this which is why it is in development.xml. That means we can discuss varying it if there is something flawed in the approach proposed.

Igor-Misic, 2023-10-11:
Initially, I shared those thoughts. After started looking into the code I realized this kind of implementation would require way more code changes, and it would be more complex.

Currently, QGroundcontrol is sending polygons and circles as mission items in a `for loop` one after another. It doesn't care if it is mission, geofence, or relly point. 
See: https://github.com/mavlink/qgroundcontrol/blob/c002ef6f702be1a0ec52914754ab898e3f0e538b/src/MissionManager/GeoFenceManager.cc#L105

This means we would need to find the best place to send [MAV_CMD_SET_FENCE_BREACH_ACTION](https://mavlink.io/en/messages/development.html#MAV_CMD_SET_FENCE_BREACH_ACTION) in between polygons, and we will need additional logic to parse it on PX4 side. 

On the other hand, a simple solution of filing one of the "reserved" parameters will solve this.

Another thing is that each action needs to be saved somewhere in PX4. We are saving polygons and circles in Dataman. To store `fenceAction` is best to store it somewhere with its own polygon or circle. Yes, there will be redundant data for each vertex, but it is the same situation with **inclusions**. 
[image]

Regarding resources in Dataman even though inclusion is bool, it will use 2 bytes of data since it is stored as `nav_cmd`. I was thinking of using 1 byte for `fence_action`.  This could be avoided if the first vertex was storing additional data (inclusion, fence_action, inclusion_group, etc.)  and every other vertex only required data. But that will be traded with additional complexity, so I will not enter too much into this now. 

Maybe param7 can be reserved for altitude and we can use **param3** for `fence_action`. 
Param2 was also reserved before  **inclusion_group** feature was added:
https://github.com/mavlink/mavlink/pull/1510

hamishwillee, 2023-10-12:
Prior art, easier development, and lower memory consumption are all good arguments - thanks for the explanation. I think we'll be OK with this approach until we run out of reserved bits :-). 

The existing approach in development.xml has not been implemented. I'm happy to take two PRs that:
a) remove the current suggestion
b) add the new update.

Do you plan to provide an implementation? We don't merge  to common (which the second "b" PR would have to do) without prototypes in place.

Igor-Misic, 2023-10-14:
Thanks a lot. Started the topic to see what its current state is so we can align. 

I am doing feasibility development at the moment at QGC and PX4. So far it looks good. Only PX4 has a very bad solution for grouping (that will need to be reworked) or removed when fence action for each fence is introduced. 

[image]

So if understand this correctly I should add messages:
MAV_CMD_NAV_FENCE_POLYGON_VERTEX_INCLUSION
MAV_CMD_NAV_FENCE_POLYGON_VERTEX_EXCLUSION
MAV_CMD_NAV_FENCE_CIRCLE_INCLUSION
MAV_CMD_NAV_FENCE_CIRCLE_EXCLUSION

with my suggested changes do development.xml (they are missing from development at the moment)? 

One more thing, can we replace the order of `FENCE_ACTION_TERMINATE` and `FENCE_ACTION_LAND`? 
https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml#L367-L372

In that case, we can have fences inside one another and a higher number will have higher severity. This would reduce logic inside PX4 to track what fence was breached and we can execute one with the highest severity. 
In this case, I think terminating the flight has the highest severity since the vehicle will fall down.

hamishwillee, 2023-10-15:
Hi @Igor-Misic 
I'm heading off to the dev day now, so here is my quick notes. What I am proposing is:
1. A PR to remove the current proposed command for setting fence actions from development.xml. THOUGH, it might be useful to keep it as a command-protocol only command to set the default fence action, rather than using parameters.
2. A PR to modify the commands you suggest IN COMMON.xml. It would obviously be better to do in development.xml but the toolchain doesn't allow you to override commands as it should. Unfortunately we can't merge that until it is accepted, but we can kind of "pre approve it".
3. > One more thing, can we replace the order of FENCE_ACTION_TERMINATE and FENCE_ACTION_LAND?
   I don't **think** so. I think this was copied from ArduPilot. But it might be that they don't use those values, in which case you could probably verify that, and then "maybe".
   
   Hope that helps.

Igor-Misic, 2023-10-16:
> A PR to remove the current proposed command for setting fence actions from development.xml. THOUGH, it might be useful to keep it as a command-protocol only command to set the default fence action, rather than using parameters.

In my implementation, I even removed the parameter with geofence action. It unreasonably complicates code while fence action is clearly arriving from QGC. I think 

> A PR to modify the commands you suggest IN COMMON.xml. It would obviously be better to do in development.xml but the toolchain doesn't allow you to override commands as it should. Unfortunately we can't merge that until it is accepted, but we can kind of "pre approve it".

Okay, I'll open PR with those suggestions. 

> I don't think so. I think this was copied from ArduPilot. But it might be that they don't use those values, in which case you could probably verify that, and then "maybe".

I had a quick look into Ardupilot and I don't think they are using it. It is not really mandatory for PX4 ATM since PX4 is using its own definitions of actions. I just wanted to see if we can reduce some complications in the future if someone decides to use those instead of PX4 having its own. 

I am good with those answers I'll proceed with actions.

closed 2023-10-16 by Igor-Misic