mavlink/mavlink #2045: Wireshark: WLua script has C stack overflow error.

bluems, 2023-10-10:
For now, I've been working on commit b295e70 on the master branch.

As of that version, I have the following issues

[Lua Error: C:\Program Files\Wireshark\plugins\mavlink_2_common.lua:42813: C stack overflow]

My guess is that there is a problem with an empty buffer or something else, but I don't know how to create a development environment and test the lua script.

hamishwillee, 2023-10-10:
What are you doing when you get that error, and can you reproduce it on the latest master? I.e. does it always happen as soon as you start wireshark?

hamishwillee, 2023-11-01:
Closing - I haven't had this error. If you are able to provide the requested information and repost as an issue in https://github.com/ArduPilot/pymavlink we can get someone to look at it.

closed 2023-11-01 by hamishwillee