mavlink/mavlink #2052: Clarification on FTP burst protocol

julianoes, 2023-11-03:
As I've been re-implementing and debugging MAVLink FTP I stumbled on the size field in a `BurstReadFile`.
BurstReadFile is the message to request a burst after the initial "open file readonly" step which 

The docs (https://mavlink.io/en/services/ftp.html#reading-a-file-burstreadfile) state that in BurstReadFile, the `size` is 4 and therefore `data` contains an int32 containing the "length of the burst", e.g. the full file, or part of it.

To me this makes sense, because it allows you to do a burst for part of the file, in case only one part is required, or only a part has been missed by the last burst.

However, looking at implementation, it looks like often `size` is set to the burst size, so max the 239 bytes available for the FTP payload. This allows tweaking the burst packet size, if required, but wouldn't allow partial file download, if I understand that correctly.

Here is the implementations that I found:

- MAVProxy: burst_size is 1..239, default 80:
https://github.com/ArduPilot/MAVProxy/blob/master/MAVProxy/modules/mavproxy_ftp.py#L318

- MissionPlanner: default looks like 80 as well:
https://github.com/ArduPilot/MissionPlanner/blob/f57bc79feba124eb9dde5c907f968c9bf2b2fdcf/ExtLibs/ArduPilot/Mavlink/MAVFtp.cs#L1473

- QGC: `sizeof(data)`, so 239:
https://github.com/mavlink/qgroundcontrol/blob/39b345e63601b0666ffe20a1aa6f7cd85cd9e077/src/Vehicle/FTPManager.cc#L315

- ArduPilot: seems to assume a `max_read` param per read
https://github.com/ArduPilot/ardupilot/blob/1bf7c9ee776797e160e75295da430b33977db9df/libraries/GCS_MAVLink/GCS_FTP.cpp#L476

- PX4: seems to just ignore the field in this case
https://github.com/PX4/PX4-Autopilot/blob/a989e5338ce3902dd63e5da41d99931d8185fdf8/src/modules/mavlink/mavlink_ftp.cpp#L1111

- MAVSDK: puts the file size in `data` as the docs say:
https://github.com/mavlink/MAVSDK/blob/578ec76a3ebc32cd269a417cf0b35279f22e5dbb/src/mavsdk/core/mavlink_ftp_client.cpp#L615-L616

So, this looks like the path of least resistance is to update the docs to reflect what is implemented in most implementations, and fix MAVSDK accordingly.

However, if I understand it correctly, this means a burst is always for a full file, and can't be used for partial downloads (e.g. to patch download holes).

I would very much appreciate any comments @hamishwillee @auturgy and @tridge. Thanks!

tridge, 2023-11-03:
@julianoes we should lock in the current implementation practice of having that field as the size of each piece of the response, and always whole file. Then we should add a new MAVFTP command for partial file burst. That will need both the fragment size and total size.
The reason we can't just use 239 is some transports will not carry a large packet. The RFD900x v1 is the worst, and it is the reason we default to 80 (which for most transports is much too small). Any more than 80 and you can't use MAVFtp at all on some RFD900x radio firmwares.

julianoes, 2023-11-04:
Got it, makes sense, thanks @tridge.

@hamishwillee we have to fix the docs then. Let's see who is quicker :)
And I have to fix the MAVSDK implementation.

I'll leave this issue open until we have these two points fixed. A partial burst would be nice, I will leave that to @tridge.

hamishwillee, 2023-11-08:
@julianoes It will be you, because I don't understand what is going on. 

My understanding (docs) is that in a burst the client  indicates the size of the `data` type using `size=4` then in the `data` it indicates the amount of data in the whole burst.

What I think you are saying is that 
- implementations are ignoring the `data` field and using the `size` to contain the size of the burst data. 
- However the since `size` is one byte this means the max value is 255.
- Which means that this value can't be treated as the "whole burst size". It has to be perhaps the "default burst size", and you just guess the total burst from the initial response to opening  the file?
- So you can't do partial file downloads because there is no way to specify the total size of the burst you want in the actual burst. Is that right?

So the change is to indicate that you can only burst read from an offset to the end of the file, because that's the information you can supply to the server?

julianoes, 2023-11-09:
> My understanding (docs) is that in a burst the client indicates the size of the data type using size=4 then in the data it indicates the amount of data in the whole burst.

That's what the doc say but that's not what is implemented anywhere, except in MAVSDK which I implemented based on the docs.

Yes, exactly all correct. And it's not just the fact that it's only one byte and limited to 255 but the meaning of it - as implemented - is to signal burst packet size.

hamishwillee, 2023-11-15:
FYI @tridge This was updated in docs to match the actual behaviour in https://github.com/mavlink/mavlink-devguide/pull/519.

If someone wants to use bursts for a partial file then they will need to propose messages and do an implementation etc.

closed 2023-11-15 by hamishwillee