mavlink/mavlink #2061: Documentation loop

AndKe, 2023-12-04:
[image]

[image]

https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_ROI

Please observe that the command is deprecated, and replaced by itself.
I was checking if the altitude parameter could be "-1" - to be ignored so that only yaw was affected by ROI - please follow the blue text on parameters 5,6 or 7   - it does not provide any additional information.

hamishwillee, 2023-12-07:
No, this says `DEPRECATED: Replaced by MAV_CMD_DO_SET_ROI_*`. That means messages that start with `MAV_CMD_DO_SET_ROI_`. So MAV_CMD_DO_SET_ROI_LOCATION, MAV_CMD_DO_SET_ROI_WPNEXT_OFFSET, and so on.

Would it help to change this to: `DEPRECATED: Replaced by separate messages starting with MAV_CMD_DO_SET_ROI_, such as MAV_CMD_DO_SET_ROI_LOCATION`.  ?

W.r.t. to the values, I'd interpret them in terms of the new messages - e.g. for WPNEXT see what the param 7 values says for https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_ROI_WPNEXT_OFFSET

Does that answer?

PS Sorry for delay, only work Weds, Thurs.

hamishwillee, 2026-08-25:
This is OK. Closing.

closed 2026-08-25 by hamishwillee