mavlink/mavlink #2062: Proposed addition/clarification of units for EPH, EPV, Azimuth

shancock884, 2023-12-04:
I was looking into doing an update to the Wireshark plugin to show the units of the data being displayed, and in the case of some particularly obscure units, to show a 'human readable' number alongside the raw value.
And this led me to notice a couple of things with the units that I think could be improved:

1/ EPH & EPV values have no unit set.
The eph and epv fields of GPS[2]_RAW_INT, HIL_GPS messages are (according to description) "unitless * 100", but there is no unit defined to be able to decode the value.
I would propose using a new unit of "1E2" - as "1" seems to be an accepted symbol for "unitless" (at least according to a quick internet search!).

Note that the eph and epv fields of HIGH_LATENCY2 do have units of "dm" defined. I guess that either the unit is incorrectly set (and it should again be some multiplier of 'unitless'), or it is misnamed, and it actually representing position uncertainly.

2/ Satellite Azimuth unit doesn't consider scale factor
The satellite_azimuth field of the GPS_STATUS message has a unit of "deg", but the description says "0: 0 deg, 255: 360 deg".
To be consistent with other units, and allow automatic value decoding, I would propose that a specific unit is used here too.
e.g. "degS8" (meaning "Degree Scaled to 8 bit integer"), or "deg/360*255" or any sensible simplification thereof.

I would also question if 255 is really supposed to be 360deg - as this means a value of 0 and 255 represents the same angle.
Perhaps what was intended is that 255 represents 360-(360/256) = 358.59375 ?

3/ Minor typo in schema comment
The mavschema.xsd file includes a comment next to the degE5 and degE7 units as "degrees * 10E5" and "degrees * 10E5".
To be fussy, this should either be written as "degrees * 10^5" (i.e. "^" not "E"), or "degrees * 1E5" (i.e. "1" not "10").
I know this is very pedantic and only applies in a comment, so I've only included it here as I thought it could be cleaned up alongside addition of proposed new units in case one or more of the above is taken forward!

hamishwillee, 2023-12-07:
@shancock884 Thanks very much for this. 

3. Yes - https://github.com/ArduPilot/pymavlink/pull/884
2. Can you PR degS8 into the xmlschema and cross link to this. That's a good place for the discussion. Then we can update the units on the field. I'm OK with 360 and 0 being the extremes of the range.
1. The value is unitless because it has no units. 
   - Raw values are normally between 0 and 20, so I assume "dm" is being used to indicate that this might be sent as a value between 0 and 200, fitting nicely into a uint8. But as you say, it is "wrong". 
   - I'm not sure what to do with these cases because a scaling factor is not a unit. I'll add for discussion in next mav call - 2 weeks.

hamishwillee, 2023-12-13:
Thanks for that. I added this to next week's agenda https://github.com/mavlink/mavlink/wiki/20231221-Dev-Meeting (actually on Weds 20th not 21st).FYI @peterbarker @auturgy @julianoes

auturgy, 2023-12-20:
We can't really change the definition of the range (it would result in an undefined range between 255 and 0).

shancock884, 2023-12-20:
@hamishwillee - Julian looked at implementation of eph and epv fields of HIGH_LATENCY2. On PX4 these do represent 'position uncertainly', and so the existing unit of "dm" is valid, and on Ardupilot they are hard-coded to 0 at present.
This means that there is no need for a "unless E1" or "1E1" unit - only the "unitless E2" or "1E2" one for GPS[2]_RAW_INT and HIL_GPS messages.

The confusion comes from the fact that the fields called eph/epv on the GPS messages are really misnamed (should have been hdop/hvop), and don't mean the same thing as the 'correctly named' eph/epv fields of HIGH_LATENCY2.

hamishwillee, 2023-12-21:
Thanks @shancock884 - created in https://github.com/ArduPilot/pymavlink/pull/894. Let's see how it goes

PR #2120 (2024-05-29): common - multiplier values for azimuth and eph/epv
We added a `multiplier` attribute to indicate that the value is scaled. The value is the amount you multiply the received value to get the actual value in the specified units (if there are any).

This adds multiplier for eph/epv of 1E-2 and multiplier of `360/255` for satellite_azimuth

Fixes #2062

Related to https://github.com/ArduPilot/pymavlink/pull/885, https://github.com/ArduPilot/pymavlink/pull/894

Docs update in https://github.com/mavlink/mavlink-devguide/pull/540

@shancock884 @peterbarker @julianoes This OK?
closed 2024-05-29 by hamishwillee