mavlink/mavlink #2073: MAVGEN LUA WIRESHARK cannot handle negative lattitudes/longitudes

PeterJBurke, 2024-01-14:
I have created the LUA script to work with Wireshark to parse mavlink UDP packets, 
using the directions here:
https://mavlink.io/en/guide/wireshark.html
Specifically, I run the command: 
python3 -m pymavlink.tools.mavgen --lang=WLua --wire-protocol=2.0 --output=mavlink_2_common message_definitions/v1.0/common.xml
It creates the lua script "mavlink_2_common.lua" which I then incorporate with Wireshark.
It all works fine, but when there is a negative latitude or longitude, it shows the bits correctly but a zero for the values.
The screenshot shows the 4 bytes for the lat in blue are correctly displayed as the lat , but the next four bytes show zero for the lat above. (The actual lat is -117... if you check the 4 bytes).
So it must be something about signed vs unsigned int32 in Lua/Mavlink.
[image]

shancock884, 2024-02-07:
This sounds like it might be the same issue as #1862, which was cross-posted to https://github.com/ArduPilot/pymavlink/issues/701, as generators are part of pymavlink module.

Reading those previous issues, it seems that no-one got to the bottom of it at the time, but one hypothesis was that this was specific to Mac.... can you confirm what platform you are using (your screenshot certainly looks like a Mac!), and also what version of pymavlink (i.e. mavgen) and Wireshark you are using.

With Wireshark v4.2 and pymavlink 2.4.41, I can certainly see negative values on my Linux system.

shancock884, 2024-04-25:
@rotu It looked from the screenshot in https://github.com/mavlink/mavlink/issues/2105 that you are on Mac, right? I assume you can see negative lat/lon values ok?!

rotu, 2024-04-25:
> @rotu It looked from the screenshot in https://github.com/mavlink/mavlink/issues/2105 that you are on Mac, right? I assume you can see negative lat/lon values ok?!

No. This appears to be a regression and I can see the impact on Mac in the tests as well:

```console
pytest -vv
...
tests/snapshottests/test_wlua.py::test_wlua[common.xml-gps_global_origin.pcapng] FAILED       [ 98%]
tests/snapshottests/test_wlua.py::test_wlua[common.xml-mission_item_int.pcapng] FAILED        [100%]                     
...
__________________________ test_wlua[common.xml-gps_global_origin.pcapng] ___________________________
...
E       assert [+ received] == [- snapshot]
E           dict({
E                 ...
E                       latitude (int32_t) [degE7]: 517015586 (51.7015586 deg)
E         -             longitude (int32_t) [degE7]: -3332886 (-0.3332886 deg)
E         +             longitude (int32_t) [degE7]: 0 (-0.3332886 deg)
E                       altitude (int32_t) [mm]: 56990
E                 ...
E           })
...
___________________________ test_wlua[common.xml-mission_item_int.pcapng] ___________________________
...
>       assert props_to_match == snapshot()
E       assert [+ received] == [- snapshot]
E           dict({
E                 ...
E                       x: Latitude (int32_t): 514775152
E         -             y: Longitude (int32_t): -25939161
E         +             y: Longitude (int32_t): 0
E                       z: Ground Altitude (float) m: 0
E                 ...
E           })

```

shancock884, 2024-04-25:
Ok thanks for checking that.... Very odd!
Not sure if its a regression or always been like that on mac, as https://github.com/mavlink/mavlink/issues/1862 first mentions it nearly 2 years ago.
I can try to make a minimum LUA+PCAP combo to try out on Windows/Linux/Mac, and if need-be will create an issue on Wireshark's Gitlab.

rotu, 2024-04-25:
Guessing it's this line: https://gitlab.com/wireshark/wireshark/-/blob/259bbbff2a5257319ff69aa32cc377fc501eb022/epan/wslua/wslua_tree.c#L535

```c
case FT_INT32:
  item = proto_tree_add_int(tree_item->tree,hfid,tvbr->tvb->ws_tvb,tvbr->offset,tvbr->len,wslua_checkuint32(L,1));
  break;
```

And that it's supposed to be `wslua_checkint32` instead of `wslua_checkuint32`.

shancock884, 2024-04-26:
Good spot - yes that looks like it could be the culprit!
If I am following it correctly, the incoming number gets cast to an uint32, then this is passed as an argument to a function expecting an int32.
I guess that somewhere in this lot, the behaviour between Mac and Linux differs, and the negative number gets cut at 0.
Maybe the equivalent of something like this (which prints -100 on Linux):
```
#include <stdio.h>
void func(int v) {
  printf("%d\n",v);
}
int main(void) {
  double x = -100;
  unsigned int y = (unsigned int)x;
  func(y);
  return 0;
}
```

In the meantime, a fix on pymavlink side would be to remove the value argument from calls to tree:add_le, but we do need to be a little careful with this, as there are cases where we need it. For example when a float command parameter represents an enum, we need to use the value, as otherwise Wireshark would incorrectly interpret the float bytes as a raw uint32.

rotu, 2024-04-26:
> I guess that somewhere in this lot, the behaviour between Mac and Linux differs, and the negative number gets cut at 0.
> Maybe the equivalent of something like this (which prints -100 on Linux):
> ```c
>   double x = -100;
>   unsigned int y = (unsigned int)x;
> ```

Honestly, I thought that was a ridiculous suggestion and there's no way that differs between Linux and Mac. ...but it turns out you're right! It does indeed print 0 on my computer when compiling with both clang 15 and gcc 13!

> In the meantime, a fix on pymavlink side would be to remove the value argument from calls to tree:add_le, but we do need to be a little careful with this, as there are cases where we need it. For [...]

It might also make sense to use `ftypes.DOUBLE` for numeric values but I think that changes the semantics of `valuestring` in such a way that it no longer describes enum values.

BTW submitted merge request at https://gitlab.com/wireshark/wireshark/-/merge_requests/15438

shancock884, 2024-04-27:
I had a crack at a quick workaround/fix on the pymavlink side in this branch: https://github.com/shancock884/pymavlink/tree/mac-neg-int-fix
It removes the 'value' arg from calls to add_le, unless there is an associated enum - to make sure it doesn't break the case of a double or float being linked to an enum, as is the case on a number of command messages.
Its not very optimised yet, as 'value = tvbrange:...' line is still generated, even when not needed, but should prove the concept.

rotu, 2024-04-28:
It would also suffice to promote integers to int64.

```lua
f.cmd_MAV_CMD_NAV_VTOL_LAND_y = ProtoField.new("y: Longitude (int32_t)", "mavlink_proto.cmd_MAV_CMD_NAV_VTOL_LAND_y", ftypes.INT64, nil)

-- ...

    value = tvbrange:le_int64()
    subtree = tree:add_le(f.cmd_MAV_CMD_NAV_VTOL_LAND_y, tvbrange, value)
```

That said, I don't like this code at all :-(.

ksols, 2026-10-04:
I reproduced and bisected this on an Apple Silicon Mac. It was a Wireshark bug, and it has been fixed on both the Wireshark and the pymavlink side.

**Cause:** in Wireshark ≤ 4.2.4, `TreeItem:add_le()` converted the value for `FT_INT32` fields with `wslua_checkuint32()` instead of `wslua_checkint32()`. Converting a negative double to an unsigned integer is undefined behaviour in C; on Mac it produced 0. That's why the raw bytes and the `(... deg)` text (computed in Lua) were correct while the int32 field itself showed 0.

**Fixes:**
- **Wireshark 4.2.5+:** fixed by [15392c32](https://gitlab.com/wireshark/wireshark/-/commit/15392c324d5eaefcaa298cdee09cd5b40b12e09c) "wslua: Fix integer validation signedness" (backported to 4.2 as 4dadd9b0, the only relevant change to `wslua_tree.c` between 4.2.4 and 4.2.5).
- **pymavlink:** ArduPilot/pymavlink@052a201a (@shancock884, 27 April 2024) stopped passing the value to `add_le` for plain integer fields, which avoids the bug even on older Wireshark versions.

**Test** (tshark, using `gps_global_origin.pcapng` / `mission_item_int.pcapng` from pymavlink's snapshot tests):

| Wireshark | Generator before 052a201a | Current generator |
|---|---|---|
| 4.2.4 | ❌ `0` | ✅ |
| 4.2.5, 4.2.6, 4.2.14, 4.4.0, 4.6.4 | ✅ | ✅ |

With Wireshark 4.2.4 and the old generator:
```
longitude (int32_t) [degE7]: 0 (-0.3332886 deg)
y: Longitude (int32_t): 0
```
With Wireshark 4.2.5:
```
longitude (int32_t) [degE7]: -3332886 (-0.3332886 deg)
y: Longitude (int32_t): -25939161
```

Upgrading either Wireshark (≥ 4.2.5) or pymavlink (any release after April 2024) fixes it. I think this issue and #1862 can be closed (the pymavlink cross-post, ArduPilot/pymavlink#701, is already closed).

closed 2026-10-07 by hamishwillee