mavlink/mavlink #2076: mavlink packet encryption

AhmedMohanad, 2024-02-04:
hello, i have a task (encrypt the payload in mavlink packet) my question is in code here can i find the payload or the packet data and encrypt it befor send

auturgy, 2024-02-04:
The usual approach is to encrypt the transport, not the raw stream. Can you explain why this method won't work for you?

AhmedMohanad, 2024-02-04:
i ask about the part of code that cary the information section in packet
called (payload) that cary the data . where i can find it in code of python

AhmedMohanad, 2024-02-04:
and who to encrypt th transport

hamishwillee, 2024-02-07:
What is being suggested here https://github.com/mavlink/mavlink/issues/2076#issuecomment-1925903182 is that you don't encrypt the MAVLink packets yourself in code. 
Instead you set up MAVLink to run over an encrypted radio link such as provided by a secure radio. That way you don't change anything in MAVLink at all.

MAVLink does not support encryption of the payload at the packet level and I don't recommend you do this yourself because secure systems are hard to get right.

I'm setting this closed as "not supported".

closed 2024-02-07 by hamishwillee

hamishwillee, 2024-02-07:
> https://github.com/ArduPilot/mavlink/pull/349/files#r1482124891

No. I'd like to, but what you're asking isn't supported, so doing it would require lots of time, effort and support. For a start, you can't just do it for a flight stack, you have to do it end to end - updating any routers and your ground station. And that means you need to use C rather than Python, because that's what the flight stacks are written in usually.

If you want a view on how someone might do this in C you can look at https://github.com/mavlink/mavlink/pull/1334 and https://github.com/PX4/PX4-Autopilot/issues/13538. You will note that this is not merged because even though it "works" there are lots of concerns and tradeoffs. It has never been deployed.

If you're determined to do this then you would need to modify the Pymavlink parser to send out/received packets that have a new [incompatibility bit](https://mavlink.io/en/guide/serialization.html#incompat_flags) defined to indicate that the packets are encrypted. Then you'd actually have to define a certificate/key infrastructure, and modify parser to modify incoming/outcoming packets to be decrypted/encrypted. I don't know how to do that - you'd have to look at the existing code and work it out.

AhmedMohanad, 2024-02-08:
> <https://github.com/mavlink/mavlink/pull/1334> and PX4/PX4-Autopilot#13538

thank you for help. I will do my best