mavlink/mavlink #2093: GIMBAL_DEVICE_FLAGS yaw_lock and yaw_in_vehicle/earth_frame are unnecessarily redundant

Davidsastresas, 2024-03-11:
From common.xml definition of GIMBAL_DEVICE_FLAGS:

```
GIMBAL_DEVICE_FLAGS_YAW_LOCK=16, /* Lock yaw angle to absolute angle relative to North (not relative to vehicle). If this flag is set, the yaw angle and z component of angular velocity are relative to North (earth frame, x-axis pointing North), else they are relative to the vehicle heading (vehicle frame, earth frame rotated so that the x-axis is pointing forward). | */
   GIMBAL_DEVICE_FLAGS_YAW_IN_VEHICLE_FRAME=32, /* Yaw angle and z component of angular velocity are relative to the vehicle heading (vehicle frame, earth frame rotated such that the x-axis is pointing forward). | */
   GIMBAL_DEVICE_FLAGS_YAW_IN_EARTH_FRAME=64, /* Yaw angle and z component of angular velocity are relative to North (earth frame, x-axis is pointing North). | */
   
```

depending on the value of GIMBAL_DEVICE_FLAGS_YAW_LOCK it affects what GIMBAL_DEVICE_FLAGS_YAW_IN_VEHICLE_FRAME and GIMBAL_DEVICE_FLAGS_YAW_IN_EARTH_FRAME should represent. So I think according to this definition, GIMBAL_DEVICE_FLAGS_YAW_LOCK is redundant.

From my point of view I think YAW_LOCK flags should only represent if the gimbal is in yaw lock or yaw follow behaviour, and I think we should leave exclusively YAW_IN_VEHICLE/EARTH_FRAME to represent the yaw frame.

This would also simplify implementations in all ends.

For awareness, @julianoes @rmackay9 @hamishwillee 

Thanks!

julianoes, 2024-03-11:
@Davidsastresas I think you're correct that it is redundant, however, that's because it is meant to be backwards compatible.

Initially, we would transmit the yaw angle relative to North if yaw lock was enabled, and relative to forward if yaw lock was disabled. However, that was confusing, and not always easy to achieve, as some gimbals can easily send one but not the other. Therefore, it is now up to the gimbal to report whichever it wants and signal it using one of the two flags.

Davidsastresas, 2024-03-11:
@julianoes thanks for the clarification, I wasn't aware of that backwards compatibility. Closing this then!

closed 2024-03-11 by Davidsastresas

hamishwillee, 2024-03-13:
@julianoes @Davidsastresas Is this captured in gimbal docs? It should be minimally there, and ideally also in the messages.

Davidsastresas, 2024-03-13:
@hamishwillee I don't think it is. I think we could add a small section somewhere in the docs to make this very obvious, I think it is easy to mess up so would be good to remark it in the docs.

How are docs managed for mavlink? Is it accesible like a repo I could make a PR against? Thanks!

hamishwillee, 2024-03-13:
@Davidsastresas Same process :-). From [the relevant page](https://mavlink.io/en/services/gimbal_v2.html) (all pages) there is a link at the top  that you can click to edit on github  (you may have to scroll up)

[image]

This can also be build locally, but you don't need that for this.