mavlink/mavlink #2105: Wireshark dissector AUTOPILOT_VERSION

dronevdl, 2024-04-24:
Lua error is dissecting the AUTOPILOT_VERSION message.

Daily build of QGroundControl, CubeOrange with PX4.

Here is the screenshot of Wireshark:
[image]

Here is a pcap file with one packet to reproduce the problem:
[autopilotversion.zip](https://github.com/mavlink/mavlink/files/15096338/autopilotversion.zip)

hamishwillee, 2024-04-25:
FYI @rotu @shancock884  - in case you can advise

rotu, 2024-04-25:
[image]

no such problem here.
Wireshark version 4.2.4
[mavlink-common.lua.zip](https://github.com/mavlink/mavlink/files/15104505/mavlink-common.lua.zip)

What version of Wireshark are you using? Can you upload your .lua?

shancock884, 2024-04-25:
This appears to be exactly the issue described in: https://github.com/ArduPilot/pymavlink/issues/852.
The issue is fixed in PR https://github.com/ArduPilot/pymavlink/pull/886, which was included in the 2.4.41 release of pymavlink (or any recent master branch).

dronevdl, 2024-04-25:
I am sorry, it works now.
I did a git clone a long time ago, did a git pull yesterday, but apparently that did not update everything.
I did  rm -rf and git clone today and now it works.
Wireshark 3.2.3 on Ubuntu 20.04.
This can be closed, sorry for the inconvenience.

shancock884, 2024-04-25:
No worries.... good that you are sorted.
It wont affect this, but for your info: on Ubuntu, the Wireshark version on the normal package repo is quite old, so if you want a more up-to-date version, you need to do: `sudo add-apt-repository ppa:wireshark-dev/stable` before installing or updating.

closed 2024-04-25 by dronevdl