mavlink/mavlink #2109: GPS_RTCM_DATA and associated ardupilot code needs to allow more than 4 fragments of RTCM data for multi-gnss on F9P

rrr6399, 2024-04-26:
Currently, there is a restriction to 4 fragments of RTCM data totaling to a maximum of 720 bytes. I am using the F9P with all constellations (GPS, Glonass, Galileo, Beidou, QZSS, SBAS) turned on. Even with MSM4 messages, the number of bytes can be 1029 or more.) MSM7 would be higher. 

We should double the size of the maximum allowed message size.

hamishwillee, 2024-05-01:
[GPS_RTCM_DATA](https://mavlink.io/en/messages/common.html#GPS_RTCM_DATA) does indeed only allow up to four fragments according to the docs of the flags. There aren't any spare bits in the flags to allow more using the same strategy.
It's also a pity the allowed fragment size wasn't bigger, since the message truncates anyway,

Assuming it is a real problem (I have no direct experience I think we'd be best creating an updated message, since I don't see any way to fix the current message without an extension, which would prevent zero-byte truncation and be far more costly than a new message. 

Is there an upper bound we can reasonably apply to the total message size this message would ever need to carry?
Are there any other flags

We could use a double size flags field in order to allow more fragment addresses. I'd be tempted though to remove the arbitrary frame limit and encode both fragment and max fragments. If we're worried about data size we could do this in one byte - 4 bits still gives us 16 fragments. Either way, unless there is a reason to restrict the maximum size per packet to 180 we should make that maximum fill the whole MAVLink packet, since there is no reason to restrict it when we can't ever extend the new message without the high cost of losing zero-byte truncation.

@peterbarker Any thoughts on this one?

peterbarker, 2024-05-04:
I also can't see extending this message without breaking compatibility.   I'll bring it up at an ArduPilot DevCall.

tridge, 2024-05-06:
@rrr6399 can you provide an example RTCM stream with msgs over 720 in length?

peterbarker, 2024-05-06:
Agree this is probably a new packet, but let's have a look at an existing packet to see if there's something clever we can do with current formwat first (which won't require everyone to update everything).

tridge, 2024-05-06:
if we did a new msg we'd need to work out how the client knows which msg to send

rrr6399, 2024-05-07:
> @rrr6399 can you provide an example RTCM stream with msgs over 720 in length?

I'll archive a few. Are you not seeing this issue with any of your F9P systems? If not, it is possible that I have a bug.

hamishwillee, 2024-05-09:
@tridge Are you seeing issues with F9P as per ^^^

Discussed briefly in the mav call last night. @auturgy noted that it is very easy to clog the link with RTCM messages, and if we decide to allow more data on the link we need to look at cost vs benefit and possible mitigations.

rrr6399, 2024-05-09:
> @rrr6399 can you provide an example RTCM stream with msgs over 720 in length?

@tridge It is possible that interspersed NMEA messages in the stream exposed a bug in my parser. I'm still looking into it. When I turn off all NMEA messages, I no longer see RTCM message that are longer than 720 at least for my bench testing near a window. I'll do an outdoor test soon as well.

codeceptsDE, 2024-06-22:
> https://github.com/mavlink/mavlink/issues/2109#issuecomment-2087873458
> Is there an upper bound we can reasonably apply to the total message size this message would ever need to carry?

RTCM Version 3 messages have a frame structure of [3 bytes header = 8bit preamble + 6bit reserved + 10bit length] + [0-1023 bytes data] + [3 bytes CRC], per the [RTCM SC-104 Standard](https://en.wikipedia.org/wiki/RTCM_SC-104#Version_3). [^1]
So the **maximum length of an RTCMv3 message** would equal **1029 bytes**.

I am surprised that 720 bytes was chosen as the limit for `GPS_RTCM_DATA` instead of this 1029 byte limit – which would seem like the natural choice. As I understand, this would allow any valid RTCMv3 message to fit.

As an aside: What has been the rationale for a 180 byte payload per `GPS_RTCM_DATA` message, is this known? As I understand, Mavlink messages can carry up to 255 bytes of payload. Unfortunately, even 4x255=1020 is still <1029.

> https://github.com/mavlink/mavlink/issues/2109#issuecomment-2087873458
> Assuming it is a real problem (I have no direct experience)

We experienced a here3 RTK (ublox M8P) exibiting very odd behavior when injecting RTCM over Mavlink for _some_ message types. My best explanation to date is that the M8P (or here3 firmware) is actually crashing/resetting due to some (partial?) RTCM messages being dropped as a consequence of this 720 byte limit. Eliminating some types from the correction data stream seems to resolve this. Another explanation would be M8P choking on RTCM messages of types it will not parse, but I doubt that.
[^1]: The actual RTCM 10403.3 standard is only accessible to RTCM members, but you still find these definitions online.

hamishwillee, 2024-06-25:
@julianoes Is this something you could look at as part of work for HERE3 RTK? @tridge Did you see this further feedback ^^^

@codeceptsDE 
1. Any reasoning on this message predates me. Perhaps it predates the RTCMv3 and there was a smaller data size in earlier versions?.
2. Yes, messages can carry up to 255 bytes. I don't know why they wouldn't have been filled.