mavlink/mavlink #2124: The attribute 'multiplier' is not allowed error

Aleksandr-Sopliakov, 2024-06-24:
The commit introducing the problem is https://github.com/mavlink/mavlink/commit/f1d42e2774cae767a1c0651b0f95e3286c587257

generating C++ dialect gives the following output:

```sh
ERROR: /home/wiz29/.cache/bazel/_bazel_wiz29/16df959b0e695745ca275009ddca6358/sandbox/linux-sandbox/275/execroot/_main/external/_main~third_party_extensions~mavlink/message_definitions/v1.0/common.xml:5221:0:ERROR:SCHEMASV:SCHEMAV_CVC_COMPLEX_TYPE_3_2_1: Element 'field', attribute 'multiplier': The attribute 'multiplier' is not allowed.
/home/wiz29/.cache/bazel/_bazel_wiz29/16df959b0e695745ca275009ddca6358/sandbox/linux-sandbox/275/execroot/_main/external/_main~third_party_extensions~mavlink/message_definitions/v1.0/common.xml:5222:0:ERROR:SCHEMASV:SCHEMAV_CVC_COMPLEX_TYPE_3_2_1: Element 'field', attribute 'multiplier': The attribute 'multiplier' is not allowed.
/home/wiz29/.cache/bazel/_bazel_wiz29/16df959b0e695745ca275009ddca6358/sandbox/linux-sandbox/275/execroot/_main/external/_main~third_party_extensions~mavlink/message_definitions/v1.0/common.xml:5240:0:ERROR:SCHEMASV:SCHEMAV_CVC_COMPLEX_TYPE_3_2_1: Element 'field', attribute 'multiplier': The attribute 'multiplier' is not allowed.
/home/wiz29/.cache/bazel/_bazel_wiz29/16df959b0e695745ca275009ddca6358/sandbox/linux-sandbox/275/execroot/_main/external/_main~third_party_extensions~mavlink/message_definitions/v1.0/common.xml:6147:0:ERROR:SCHEMASV:SCHEMAV_CVC_COMPLEX_TYPE_3_2_1: Element 'field', attribute 'multiplier': The attribute 'multiplier' is not allowed.
/home/wiz29/.cache/bazel/_bazel_wiz29/16df959b0e695745ca275009ddca6358/sandbox/linux-sandbox/275/execroot/_main/external/_main~third_party_extensions~mavlink/message_definitions/v1.0/common.xml:6148:0:ERROR:SCHEMASV:SCHEMAV_CVC_COMPLEX_TYPE_3_2_1: Element 'field', attribute 'multiplier': The attribute 'multiplier' is not allowed.
Validating external/_main~third_party_extensions~mavlink/message_definitions/v1.0/common.xml
```

Other configurations: minimal, standard works perfectly.
The interesting fact that the commit did not pass the xml validation.
[image]

hamishwillee, 2024-06-25:
Thanks @Aleksandr-Sopliakov 

I think you might need to update your submodules - this works fine for me:

```
python3 -m pymavlink.tools.mavgen --lang=C++11 --wire-protocol=2.0 --output=generated/include/mavlink/v2.0 message_defi
nitions/v1.0/common.xml
Validating message_definitions/v1.0/common.xml
Parsing message_definitions/v1.0/common.xml
Validating /home/ubuntu/github/mavlink/message_definitions/v1.0/standard.xml
Parsing /home/ubuntu/github/mavlink/message_definitions/v1.0/standard.xml
Validating /home/ubuntu/github/mavlink/message_definitions/v1.0/minimal.xml
Parsing /home/ubuntu/github/mavlink/message_definitions/v1.0/minimal.xml
Found 223 MAVLink message types in 3 XML files
Generating C headers
Generating C implementation in directory generated/include/mavlink/v2.0/common
Generating C implementation in directory generated/include/mavlink/v2.0/standard
Generating C implementation in directory generated/include/mavlink/v2.0/minimal
Copying fixed headers for protocol 2.0 to generated/include/mavlink/v2.0
Generating C++ implementation in directory generated/include/mavlink/v2.0/common
Generating C++ implementation in directory generated/include/mavlink/v2.0/standard
Generating C++ implementation in directory generated/include/mavlink/v2.0/minimal
Copying fixed C++ headers for protocol 2.0 to generated/include/mavlink/v2.0
```

The error is valid, but only affects documentation, which I have not yet updated.

hamishwillee, 2024-09-04:
Closing as this is not valid AFAIK

closed 2024-09-04 by hamishwillee