mavlink/mavlink #2155: TERRAIN_REPORT?

enok71, 2024-09-26:
I'm trying to understand the terrain protocol described in https://mavlink.io/en/services/terrain.html. I'm starting to think there is some sort of confusion in either the functionality or in the documentation:
 
The lat, lon position in a TERRAIN_REPORT is supposed to match that in a preceeding TERRAIN_CHECK if a TERRAIN_CHECK triggered it. I.e. the lat, lon is in general not the position of the drone. The terrain_height must be the terrain height at the lat, lon coordinates (not at the drone's position). On the other hand "current_height" is clearly the drone's altitude. But over what? Over the terrain_height at lat, lon? Or over the terrain_height at the drone's current position?

TERRAIN_REPORT can also be streaming back after a TERRAIN_REQUEST in parallel with the TERRAIN_DATA flow. But what lat, lon positions are then intended to be transmitted? Something that should match the position of tiles transmitted in TERRAIN_DATA messages I guess (if the purpose is to help debug the TERRAIN_DATA flow). Still very unclear though. What lat, lon, terrain_height and current_height are emitted in this case?

It seems that ardupilot streams TERRAIN_REPORT messages for a different reason, namely as a small message containing the drone's current altitude over ground. The lat, lon and terrain_height is then always those of at the drone's position. However, this use of TERRAIN_REPORT is not in accordance with the documentation at https://mavlink.io/en/services/terrain.html. Seems like a third way of using TERRAIN_REPORT not described in https://mavlink.io/en/services/terrain.html?

What is the right way of using TERRAIN_REPORT according to you? Should the ardupilot way of using it be described or at least mentioned in your documentation?

hamishwillee, 2024-10-02:
G'day, been on holiday sorry.

The information is based on this discussion https://github.com/mavlink/mavlink-devguide/pull/259#issuecomment-651316489. 
It is possible, and even likely that parts of it are wrong or inconsistent with what a flight stack might be doing.
As you say it is important to capture both expected behaviour and differences. 

> The lat, lon position in a TERRAIN_REPORT is supposed to match that in a preceding TERRAIN_CHECK if a TERRAIN_CHECK triggered it. I.e. the lat, lon is in general not the position of the drone. The terrain_height must be the terrain height at the lat, lon coordinates (not at the drone's position). On the other hand "current_height" is clearly the drone's altitude. But over what? Over the terrain_height at lat, lon? Or over the terrain_height at the drone's current position?

As you say, when a [gcs checks tile data](https://mavlink.io/en/services/terrain.html#gcs-terrain-tile-check) it sends a `TERRAIN_CHECK` with a lat/lon, and expects back a `TERRAIN_REPORT` for that lat/lon + current vehicle height. 
- I would **expect** `current_height` to be MSL over the lat, lon - but we should confirm (at end).
- Not sure about value of `pending` and `loaded` - presumably these are the values last populated when downloading terrain data.

According to that linked discussion ArduPilot and PX4 both stream `TERRAIN_REPORT` and PX4 doesn't implement `TERRAIN_CHECK` for the request. That would seem to indicate that the GCS has no way to check tile data on PX4 at least.

> TERRAIN_REPORT can also be streaming back after a TERRAIN_REQUEST in parallel with the TERRAIN_DATA flow. But what lat, lon positions are then intended to be transmitted? Something that should match [...]

Yes, [when drone is requesting map data](https://mavlink.io/en/services/terrain.html#autopilot-terrain-map-request).

My expectation is that the only interesting part of terrain report for this workflow are `pending` and `loaded` - indicating progress relative to the grid defined by the last `TERRAIN_REQUEST`. The other information could match any tile/TERRAIN_DATA - I suspect just the last one uploaded or requested - as I say, doesn't seem useful. I'll check that.

> It seems that ardupilot streams TERRAIN_REPORT messages for a different reason, namely as a small message containing the drone's current altitude over ground. The lat, lon and terrain_height is then [...]

I will need to talk to ardupilot about this. 

What I think is that if you're not implementing `TERRAIN_CHECK` it doesn't matter - the `TERRAIN_REPORT` is only really useful for getting the current status of the download - current and pending. I'll make a separate post below to highlight some questions here.

hamishwillee, 2024-10-02:
@DonLakeFlyer @bkueng @peterbarker Some questions here about the [terrain protocol](https://mavlink.io/en/services/terrain.html) on PX4 and ArduPilot. Some of these questions previously came up in https://github.com/mavlink/mavlink-devguide/pull/259 but I missed them :-(

1. Does either PX4 or ArduPilot support `TERRAIN_CHECK`.
   - If not, should it be part of the protocol?
   - If it is supported, what would be set for the pending and loaded bits in TERRAIN_REPORT ?
2. According to the poster:
   > Ardupilot streams TERRAIN_REPORT messages containing the drone's current altitude over ground. The lat, lon and terrain_height is then always those of at the drone's position.

   - @peterbarker Is this correct? What's the logic here/what do you use the TERRAIN_REPORT for with this data?
   - @bkueng In https://github.com/mavlink/mavlink-devguide/pull/259#issuecomment-665720443 you stated:
   
      > I did not implement TERRAIN_CHECK, as TERRAIN_REPORT is streamed already.
      > Also for the report, I do not fill out the lat, lon or height. I don't see the use for that, as there are other messages for that already.
      
      Is that still true?

      This makes sense if the TERRAIN_REPORT is only for reporting progress, but not if you want the GCS to be able to check whether the drone has a particular tile. If you want that functionality you have to implement the fields and TERRAIN_CHECK.

3. Is `TERRAIN_REPORT.current_height` in MSL or relative to the ground? It sounds like for ArduPilot it is AGL, while PX4 doesn't report it.
4. @DonLakeFlyer The [autopilot requests a map](https://mavlink.io/en/services/terrain.html#autopilot-terrain-map-request) and at the end it re-requests any missing tiles using `TERRAIN_REQUEST` right? In this case, my assumption is that 
   - it would only fill in the bits in the mask for the tiles that were missing
   - BUT the `TERRAIN_REPORT.loaded` would reflect the whole mask/array/grid right for ? - i.e. say you started with a request for all tiles, and received all but three of them, you'd do another `TERRAIN_REQUEST` with just those bits for the missing tiles set. But the `TERRAIN_REPORT` would initially have `pending=3` and `loaded=size of the whole array you already downloaded in the first request". (Specifically, it would NOT  be `pending=3`, and `loaded=0`). Right?

peterbarker, 2024-10-02:
>     1. Does either PX4 or ArduPilot support `TERRAIN_CHECK`.

ArduPilot does, and has for a decade :-)

>        * If it is supported, what would be set for the pending and loaded bits in TERRAIN_REPORT ?

pending is the number of requests the GCS might expect to see; graphing this over time is often useful to see if your vehicle is missing terrain which it would really like to have.

loaded is just how full the autopilot's cache is; really just diagnostic information from the autopilot.

>     2. According to the poster:
>        > Ardupilot streams TERRAIN_REPORT messages containing the drone's current altitude over ground. The lat, lon and terrain_height is then always those of at the drone's position.

That's "current_height" at the lat/lon in the message, yes.

>        * @peterbarker Is this correct? What's the logic here/what do you use the TERRAIN_REPORT for with this data?

The terrain report comes back from the autopilot.  I've been using it recently to make sure a mission I'm planning in simulation doesn't deviate too much from requested distance above terrain.  MAVProxy shows two AGL heights - one which MAVProxy calculates itself from the terrain database, and the other from this message; useful for cross-check.

>        * @bkueng In [Terrain protocol docs mavlink-devguide#259 (comment)](https://github.com/mavlink/mavlink-devguide/pull/259#issuecomment-665720443) you stated:
>          > I did not implement TERRAIN_CHECK, as TERRAIN_REPORT is streamed already.

TERRAIN_CHECK is useful if you want to get the autopilot to report data for a particular area, and for forcing the autopilot to pre-populate data for an area.  There are other/better means of making sure you have that data, of course.

>          > Also for the report, I do not fill out the lat, lon or height. I don't see the use for that, as there are other messages for that already.

Having the data all in one packet gives you a frame of data - no way to *nicely* correlate eg. GLOBAL_POSITION_INT and `TERRAIN_REPORT` messages.

>     3. Is `TERRAIN_REPORT.current_height` in MSL or relative to the ground? It sounds like for ArduPilot it is AGL, while PX4 doesn't report it.

Consider it a virtual rangefinder - it's AGL.

hamishwillee, 2024-10-02:
THanks very much @peterbarker 

Ah, so I think you are saying this (?):

TERRAIN_REPORT | - | - | -
--- | --- | --- | ---
lat	| int32_t | degE7 | Latitude of _reported tile_
lon	| int32_t | degE7 | Longitude of reported tile 
spacing |	uint16_t | m | grid spacing (zero if terrain at this location unavailable) **[APM use 0 too ?]**
terrain_height | float | m	| Terrain height MSL at lat/lon if known to vehicle **[NaN otherwise?]**
current_height | float | m	| AGL height of current vehicle altitude (irrespective of actual vehicle location), if it were at the tile location  **[NaN if tile info not available?]**
pending |	uint16_t | | Number of 4x4 terrain blocks Vehicle is waiting on from GCS **[So this is `56-loaded`?]**
loaded | uint16_t | | Number of 4x4 terrain blocks currently loaded onto Vehicle **[So this is `56-pending` ]**

I can see how this works for a TERRAIN_CHECK, but you stream it all the time. So what tile is it outputting the data for when you're satisfying a TERRAIN_REQUEST? The last tile loaded?

What about when you're neither doing a TERRAIN_CHECK OR a TERRAIN request - does it just output the tile at the current vehicle position?

hamishwillee, 2024-10-02:
Hi @peterbarker,

I don't _think_ that quite answers all my question. According to the discussion above you are streaming terrain-report _all the time_, not just in response to a terrain-check (and [according to the flow](https://mavlink.io/en/services/terrain.html#autopilot-terrain-map-request) terrain-report is streamed when you are requesting data).

So it is obvious what lat-lon will be in terrain-report in response to a terrain-check, because you specified that in the terrain-check. But if you are not doing a terrain check you didn't specify the desired lat-lon. What did you use - a random one? The current vehicle lat-lon? Same if vehicle is asking for terrain data - it might emit some value that maps to the current terrain data it sent, or some other value. 

> current height is always from the vehicle's canonical position.

Yes, but from its altitude or lat/lon. You said before "That's "current_height" at the lat/lon in the message, yes." so I am taking this to mean "the AGL altitude of the vehicle if it was at the lat/lon in the tile".

That's the only thing that makes sense to me - you want an easy way to work out if you're over the ground level of the next tile you're about to fly into.

peterbarker, 2024-10-02:
On Tue, 1 Oct 2024, Hamish Willee wrote:

> I don't think that quite answers all my question. According to the discussion above you are streaming terrain-report all the time, not just in response to a terrain-check (and according
> to the flow terrain-report is streamed when you are requesting data).

Yes, that's correct, we stream it all the time.  I fixed ArduPilot the 
other day to be able to set the report stream rate separately from the 
terrain-request stream rate.

> So it is obvious what lat-lon will be in terrain-report in response to a terrain-check, because you specified that in the terrain-check. But if you are not doing a terrain check you
> didn't specify [...]

If a GCS does a terrain-check then we'll emit a report for that location.

The streamed messages are for the current vehicle location.

>       current height is always from the vehicle's canonical position.
> 
> Yes, but from its altitude or lat/lon. You said before "That's "current_height" at the lat/lon in the message, yes." so I am taking this to mean "the AGL altitude of the vehicle if it
> was at the lat/lon in the tile".

Sorry, that field is always for the vehicle at its current location, not 
at the location in the message.

> That's the only thing that makes sense to me - you want an easy way to work out if you're over the ground level of the next tile you're about to fly into.

We did discuss this behaviour when discussing support for the ALTITUDE 
message; I can't recall what the result was, possibly extension field 
things.  I might be able dig those notes out.

hamishwillee, 2024-10-02:
Thank you. I think that is all clear now.

It is a bit naff to always use the current location of the vehicle, but it does match the name, and ardupilot probably has prior art on this API?. It also doesn't matter that much since I think @bkueng will come back and say PX4 doesn't use it - also because most of the time the tile location is the current vehicle location. 

I'll update the messages and docs tomorrow hopefully for your review. (specifically to allow the values to be NaN etc.

bkueng, 2024-10-02:
Indeed, PX4 doesn't implement it. I once implemented the protocol w/o `TERRAIN_CHECK`, but it's not used anymore either.

peterbarker, 2024-10-02:
> It is a bit naff to always use the current location of the vehicle, but it does match the name, and ardupilot probably has prior art 

Agreed, but (a) too entrenched for us to change now and (b) it is very useful to be able to plot that field and get a continuous line representing the most important number!

> I'll update the messages and docs tomorrow hopefully for your review. (specifically to allow the values to be NaN etc.

Thanks!

enok71, 2024-10-02:
Thanks for the attention and for all clarifications!

DonLakeFlyer, 2024-10-09:
TERRAIN_CHECK:  QGC doesn't use it. The assumption is that the vehicle will fail missing start if it doesn't have the terrain info it needs.

TERRAIN_REPORT: QGC has a terrain progress indicator that pops up based on the `blockLoading` and `blocksPending` values. These are the only two fields from `TERRAIN_REPORT` that affect the QGC ui. The simplest explanation is that the UI is shown when `blocksPending` is non-zero. It will then go away after a certain amount of time if there are no more blocks pending.